VOLTTRON/volttron #19: subversion clients fail due to .svn subdirectories

ChrisFreeman, 2015-02-06:
I am a developer at QualityLogic. We are working with various repository configurations, specifically trying to use [GitHub's subversion interface](https://help.github.com/articles/support-for-subversion-clients/).  The issue occurs when trying to use subversion to checkout the volttron repository. Subversion fails due to the subdirectories named ".svn" within the repository.

```bash
$ svn checkout https://github.com/VOLTTRON/volttron
A    volttron/branches
A    volttron/branches/1.x
[abridged]
A    volttron/branches/1.x/Agents/MpcAgent/src
svn: E155000: Failed to add directory '/home/chrisf/devel/subversion_test1/volttron/branches
/1.x/Agents/MpcAgent/src/.svn': object of the same name as the administrative directory
```

The failure occurs for each of the branches (vtests, 1.x, 2.x, master).

If the **.svn** content in the directory <branch>/Agents/MpcAgent/src is not needed or is obsolete can these subdirectories be renamed to .svnold or other name?

I have issued a PR for the master and 2.x branch to fix this.  I can also add PRs for vtests and 1.x.

fstshrk, 2015-02-06:
> If the .svn content in the directory /Agents/MpcAgent/src is not needed or is obsolete can these subdirectories be renamed to .svnold or other name?

Why are you not using git?

ChrisFreeman, 2015-02-06:
Valid question. Many corporations use subversion over git. Git supports
using subversion clients on git repositories (see link in OP). The question
isn't why am I not using git (which I use frequently on other projects), it
is whether VOLTTRON will support those with subversion client needs.

If git were the only approach, one could also ask why are there subversion
subdirectories in the VOLTTRON hierarchy?

On Friday, February 6, 2015, fstshrk <notifications@github.com> wrote:

> Why are you not using git?
>

jhaack, 2015-02-07:
They are artifacts from contributors checking in code developed in their own svn repo.

hashstat, 2015-02-07:
The real question is how the .svn directory got there and whether it is needed. Looking at the log for the directory, it appeared by accident when an outside contribution was added during the 1.02 push. It appears to be unneeded and I took the liberty of removing it from all branches.

> The question ... is whether VOLTTRON will support those with subversion client needs.

I don't know if the GitHub Subversion work-flow supports pull requests, but if it does not, contributing back to VOLTTRON will be difficult as it is the method we use for contributions. Please keep in mind that none of the core team uses Subversion and might not be able to assist with issues around its use.

closed 2015-02-07 by hashstat

ChrisFreeman, 2015-02-07:
Unfortunately, the branch [vtests](https://github.com/VOLTTRON/volttron/tree/vtests) also needs to have the same .svn directory removed from Agents/MpcAgent/src

craig8, 2015-02-07:
Vtests is just a temporary branch that should not be living any longer.  I will remove that branch.

ChrisFreeman, 2015-02-07:
Even Better! Thanks.

craig8, 2015-02-07:
Done.

fstshrk, 2015-02-07:
Volttron team highly recommends using git. 

Thanks

Bora