VOLTTRON/volttron #23: caching and reliability

alliao-lbl, 2015-03-12:
We have a project where we want to guarantee reliability for receiving messages sent through the Volttron message bus.  For example, if a receiving agent misses a critical message, we want to set up a way to cache messages. The recipient agent would indicate it missed the message and another agent or service could send the requested message.  Is that a capabilty built into Volttron?  Or how would you recommend implementing this functionality?

craig8, 2015-08-26:
The historians now provide caching of messages to specific points (devices/*/all, datalogger/* etc.). I think one of those would be a good starting point.

jhaack, 2017-05-26:
Closing for now. Can revisit during Message Bus enhancement

closed 2017-05-26 by jhaack