VOLTTRON/volttron #36: Implement an sMAP historian

jhaack, 2015-04-20:
Implement an sMAP based new historian for backward compatibility.

craig8, 2015-04-23:
Is the present smap driver with twisted the 2.0 version?

craig8, 2015-04-23:
Just rolling any issues that are referencing sMAP.

#12 #7 #5

jhaack, 2015-05-14:
Initial Writing Historian done.
Next the Querying half.

jhaack, 2015-05-27:
Using the POST interface to query private streams seems to have issues. To get around this, use the data, prev, or next url based lookup.

For instance /backend/api/data/uuid/<uuid>?key=<key>&starttime=1429224582000&endtime=1429225602000&limit=1000

/prev and /next work similarly and give you limit data points before or after the input time.

Can be found on: http://www.cs.berkeley.edu/~stevedh/smap2/archiver.html

craig8, 2015-09-02:
Closing this item with the above note on private stream using alterative api

PR #61 (2015-05-14): Merge from smap_historian feature branch
This merge closes #36
closed 2015-09-02 by craig8