VOLTTRON/volttron #37: Volttron logging errror

craig8, 2015-04-21:
Executing either the command
    volttron -l=/tmp/v1home.log
or
    volttron -l /tmp/v1home.log

produces the following trace back. 

Traceback (most recent call last):
  File "/home/craig8/git/volttron/env/bin/volttron", line 9, in <module>
    load_entry_point('volttron==2.0', 'console_scripts', 'volttron')()
  File "/home/craig8/git/volttron/volttron/platform/main.py", line 550, in _main
    sys.exit(main())
  File "/home/craig8/git/volttron/volttron/platform/main.py", line 489, in main
    backupCount=opts.log_backupcount)
  File "/home/craig8/git/volttron/volttron/platform/main.py", line 109, in log_to_file
    root.addHandler(handler, **kwargs)
TypeError: addHandler() got an unexpected keyword argument 'backupCount'

hashstat, 2015-04-22:
This issue was introduced by commit f4a08e0bfb9921c4916e045680bb26f2493399fc. I recommend reverting the commit and implement rotating logs in a log configuration, which is already supported.

hashstat, 2015-04-22:
Fixed by reverting the --log-backupcount option from f4a08e0bfb9921c4916e045680bb26f2493399fc and added examples/rotatinglog.py which can be customized and passed in using --log-config.

closed 2015-04-22 by hashstat