VOLTTRON/volttron #42: Agent registration and exposure through vip

craig8, 2015-04-23:
As a first cut I want to have the following standard approaches for platforms.

 1. There will only be one PlatformAgent per instance of volttron. 
 1. The single instance of PlatformAgent will have a vip_identity of 'platform'
 1. A PlatformManagerAgent can be available.  From the management console an administrator can register a vip_address.  Assuming the above two items are established then the PlatformManagementAgent will have access to all agents that have registered with the individual platforms.

fstshrk, 2015-04-23:
>    2. The single instance of PlatformAgent will have a vip_identity of
>    3. A PlatformManagerAgent can be available. From the management

Hi Craig

Just checking that we can still host multiple VOLTTRON instances on one
physical hardware system.

Thanks

craig8, 2015-04-23:
Most definatly.  The VOLTTRON_HOME just needs to be set differently for each instance

    VOLTTRON_HOME=/tmp/v1home volttron -vv
    VOLTTRON_HOME=/tmp/v2home volttron -vv

Seperate vip_addresses, but same vip_identity of 'platform'

craig8, 2015-04-30:
According to @hashstat  the "platform" name is taken for use as the actual platform.  So we thought that the name "platform.manager" would be a good alternative vip name for this.

timothykang, 2015-04-30:
If PlatformAgent is going to have the VIP identity "platform.manager", then it seems we'll need a new name for PlatformManagerAgent, to avoid confusion. Perhaps VOLTTRONCentralAgent?

craig8, 2015-04-30:
then the vip name for that agent could be volttron.central?

PR #57 (2015-05-11): Rename to VolttronCentraAgent update demo code for showing it off

Fixes #42, #41
closed 2015-05-11 by craig8