VOLTTRON/volttron #51: Agent accessible global platform settings.

kmonson, 2015-05-05:
A nice feature be to have a global registry of settings that agents could look up and set values in. 

Write privileges would be limited to platform agents, any agent could read settings.

This could be handled by the platform agent and values could be saved to disk.

A use case for this would be a global default setting for the scraping of devices by driver agents.

jhaack, 2016-10-10:
Something to re-examine with Configuration Store

kmonson, 2016-10-10:
I think the config store pretty much does this. We could certainly put a read only globally accessible configuration in the store. with everything else, but we wouldn't want to put master driver specific setting in it as in the suggested example.

craig8, 2016-10-24:
Currently these "configuration" settings are set via command line and/or in the config file.  They are querable through the Query subsystem.

This is much cleaner through the config store.  I am going to put this on the slate for 4.1 as I think it is very doable now.

jhaack, 2018-05-30:
Retrieve VIP identity of the platform an agent is on

schandrika, 2020-02-14:
This feature is provided by secure config store in 6.0 version.  Do we need to restrict read access? Currently only write access to config store can be restricted using vctl auth add (add capabilities)

shwethanidd, 2020-02-20:
We can close this as secure config store restricts write access which is sufficient

closed 2020-02-20 by shwethanidd