VOLTTRON/volttron #63: VolttronCentralAgent should be able to request to manage a PlatformAgent

craig8, 2015-05-14:
Currently a PlatformAgent calls out to the VolttronCentral agent and establishses a connection.  This will allow a PlatformAgent to be dynamically joined to the VolttronCentral agents sphere of influence.

jhaack, 2015-05-14:
If the VOLTTRON platform is hardcoded to a VOLTTRON Central it should attempt to connect to that VC and refuse requests for management from other VCs.
If the VOLTTRON platform is not hardcoded to a VC, it will respond to the first management request from a VC. THereafter, it will refuse further management requests from other VCs.
A VC may un-register a VOLTTRON platform (non-hardcoded). The VOLTTRON platform is then available for management requests from other VCs.

fstshrk, 2015-05-15:
Can a VC also issue a command to cause a VOLTTRON platform to register to
another VC.

craig8, 2015-05-15:
Sure that's not a problem since they are two different agents on two different vip networks.

closed 2015-06-16 by craig8