VOLTTRON/volttron #68: Create a historian backed by a more robust DB

jhaack, 2015-05-19:
Possibly Postgres.

craig8, 2015-05-26:
Would an of the nosql solutions be acceptable?

jhaack, 2015-05-26:
That might be another issue. This was targeted more at Postgres/MySQL which are ones we've had people specifically ask about.

fstshrk, 2015-05-26:
For time series data, a NoSQL solution may be acceptable. But a lot of the
queries we get have been about mysql/postgres.

craig8, 2015-06-09:
For postgres
 - py-postgresql package is a python 3.0 version only. 
 - psycopg2 requires pg_config (I think this is only available when postgressql is installed.)
For mysql
 - Mysql-Python is available for python 2.7 and not for python 3.0
 - Requires that mysql server (not just client) is installed.

craig8, 2015-06-17:
I will be using mysql as it is more comfortable for me.

hashstat, 2015-06-17:
@craig8, please consider sticking to a DB package complying to the [Python DBAPI](https://www.python.org/dev/peps/pep-0249/) as much as possible. This should allow easily swapping out the underlying database with any other that also supports the API.

Also, are you targeting MySQL or it's forked, open-source sister, MariaDB?

craig8, 2015-06-17:
Oh yes definitely using a DBAPI compliant driver

craig8, 2015-06-18:
On second thought I will write a SQLHistorian that does this.

closed 2015-07-31 by craig8