VOLTTRON/volttron #75: BACPipes version volttron compile on ubuntu 14.04

craig8, 2015-05-23:
Collecting BACpypes<2,>=0.10 (from volttron==2.0)
  Could not find a version that satisfies the requirement BACpypes<2,>=0.10 (from volttron==2.0) (from versions: )
  Some externally hosted files were ignored as access to them may be unreliable (use --allow-external BACpypes to allow).
No matching distribution found for BACpypes<2,>=0.10 (from volttron==2.0)

hashstat, 2015-05-26:
I am unable to reproduce the issue. I tried on Arch Linux with an existing install and on Ubuntu 14.04 with a clean clone. Perhaps it was a temporary failure? Please try again to determine if the problem remains, and if so, please provide more detail. Thanks.

craig8, 2015-06-17:
It seems that this was because the underlying system didn't have python-dev packages installed.  doing sudo pip install bacpipes on the host box brought everything needed in.  This of course brought a bunch of stuff into the system that shouldn't have needed to be there, but it works anyways.

closed 2015-06-17 by craig8

hashstat, 2015-06-24:
This has become a common error during bootstrap. Running pip as sudo is not the answer. Instead, run `env/bin/pip install bacpypes && env/bin/python bootstrap.py` if bootstrap fails.

hashstat, 2015-07-02:
Commit 9f9b3baf8285c8e96f5888dcfb1759154b490d8e fixes the underlying issue which was caused by an update in the pip package that disallowed installing from wheels when using --global-option.