VOLTTRON/volttron #83: Messages sent through the channel subsystem are dropped if the socket is closed too early

hashstat, 2015-06-08:
ZeroMQ buffers messages for sending at a later time in a different thread, however, state used to translate between local socket address and channel subsystem addresses is cleaned up immediately when the socket is either closed or garbage collected. Any messages in the queue sent after the state is cleaned up is dropped and not delivered. This provokes a race condition in the sending of data. The current solution is for peers to coordinate a socket close, but it would be nice if something more automatic could be done.

Perhaps delaying the removal of state until some later time would be most appropriate.

craig8, 2015-06-17:
@hashstat status of this?

jhaack, 2017-05-26:
@shwethanidd @craig8 

Installing agent issue fixed. Is this still a problem outside that?

jhaack, 2018-05-30:
OBE

closed 2018-05-30 by jhaack