VOLTTRON/volttron #86: AttributeError: 'AsyncResult' object has no attribute 'link'

timothykang, 2015-06-09:
Encountered on `PlatformHistorianAgent` start:
```
Traceback (most recent call last):
  File "~/volttron/env/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
    result = self._run(*self.args, **self.kwargs)
  File "~/.volttron/agents/bba29e1b-6510-495f-811b-15257baad78c/platform_historianagent-0.1/platform_historian/agent.py", line 98, in starting
    self.vip.pubsub.subscribe('pubsub', '/platform', self.__platform)
  File "~/volttron/volttron/platform/vipagent/subsystems/pubsub.py", line 278, in subscribe
    result.link(finish)
AttributeError: 'AsyncResult' object has no attribute 'link'
<Greenlet at 0x1090a0410: <bound method SQLiteHistorianAgent.starting of <platform_historian.agent.SQLiteHistorianAgent object at 0x109092810>>(<volttron.platform.vipagent.core.Core object at 0x)> failed with AttributeError
```

craig8, 2015-06-09:
This issue happens when attempting to register a listener (i.e. subscribe) to the topic /platform on the pubsub peer.  

Currently the work around is to make sure that there is a platform agent running on the volttron instance before starting the historian service.  Then the code in question should not get triggered.

hashstat, 2015-06-09:
Fixed by 31308042f8367cbf1645cd7d07dd004c3d9636ab.

commit 31308042 (2015-06-09): AsyncResult should use rawlink instead of link. Fixes #86.
closed 2015-06-09 by hashstat