VOLTTRON/volttron #91: Error interpreting timestamps with sqlite3 in Python 2.7.3

jhaack, 2015-06-09:
The sqlite3 version in Python 2.7.3 cannot deal with timestamps containing a timezone. It tries to interpret the microseconds with the tz still attached which cannot be converted to an int.

Basically, in dbapi2.py it's doing this:
microseconds = int("558982+00:00")

When it should do this (as it does in Python 2.7.6):
microseconds = int('{:0<6.6}'.format("558982+00:00".decode()))

We need to find a way to update the sqlite3 version for older Python versions. One possible solution is: 
http://stackoverflow.com/questions/14541869/how-to-upgrade-sqlite3-in-python-2-7-3-inside-a-virtualenv

jhaack, 2015-06-09:
Confirmed, changing the line in dbapi2.py to use the line from the newer version fixes the problem.

craig8, 2015-09-24:
use sqlite3.version information for detecting and hopefully handling this issue.

jhaack, 2015-10-28:
This issue is caused by: https://bugs.python.org/issue19065

jhaack, 2015-10-28:
Documented in wiki. Creating a new issue to make this work for Python 2.7.3

closed 2015-10-28 by jhaack