VOLTTRON/volttron #101: Add ability to save settings at the VC level

craig8, 2015-06-15:
This allows the arbitrary setting of a key to a value at the VC level.  This will help to allow the storage of different 'widgets' within the ui.

craig8, 2015-06-15:
Example code for console.
{
    "method": "platforms.uuid.6c448da6-92e7-448f-a8da-8af529002e6e.set_setting",
    "params": {"key": "stuff", "value":"Good"},
    "authorization": "8e6198dd-c3d2-4cc4-bd06-2b3a245f1315"
}

{
    "method": "platforms.uuid.6c448da6-92e7-448f-a8da-8af529002e6e.set_setting",
    "params": {"key": "stuff", "value":"Good"},
    "authorization": "8e6198dd-c3d2-4cc4-bd06-2b3a245f1315"
}

craig8, 2015-06-17:
Settings should be stored in $VOLTTRON_HOME/data/platform.agent.settings file.  Which they aren't currently.

jhaack, 2016-10-10:
Charts can be stored

commit 7ce0c915 (2015-06-17): Fixed #101
closed 2016-10-10 by jhaack