VOLTTRON/volttron #106: Allow VOLTTRON to run on windows

jhaack, 2015-06-18:
Make changes necessary to allow the platform to run on Windows. Note: not recommended for deployment.

fstshrk, 2015-06-18:
I hope the priority of this task is low.

hashstat, 2015-06-18:
@fstshrk Of course it is. We wouldn't be working on it if we weren't sitting bored in the airport. ;-)

craig8, 2015-06-18:
GRIN @fstshrk and @jhaack

hashstat, 2015-06-18:
TODO

* Implement gevent_readlines() for windows where select() fails on non-socket file descriptors.
* Disable encryption on the local TCP socket.

jhaack, 2015-06-18:
Platform starts but cannot start agent due to:

2015-06-17 22:45:12,165 (listeneragent-0.1 7412) <stderr> ERROR:   File "C:\User
s\user\git\volttron\env\lib\site.py", line 703, in <module>

2015-06-17 22:45:12,181 (listeneragent-0.1 7412) <stderr> ERROR:     main()

2015-06-17 22:45:12,181 (listeneragent-0.1 7412) <stderr> ERROR:   File "C:\User
s\user\git\volttron\env\lib\site.py", line 672, in main

2015-06-17 22:45:12,181 (listeneragent-0.1 7412) <stderr> ERROR:     paths_in_sy
s = removeduppaths()

2015-06-17 22:45:12,181 (listeneragent-0.1 7412) <stderr> ERROR:   File "C:\User
s\user\git\volttron\env\lib\site.py", line 124, in removeduppaths

2015-06-17 22:45:12,181 (listeneragent-0.1 7412) <stderr> ERROR:     dir, dircas
e = makepath(dir)

2015-06-17 22:45:12,181 (listeneragent-0.1 7412) <stderr> ERROR:   File "C:\User
s\user\git\volttron\env\lib\site.py", line 97, in makepath

2015-06-17 22:45:12,181 (listeneragent-0.1 7412) <stderr> ERROR:     dir = os.pa
th.abspath(dir)

2015-06-17 22:45:12,181 (listeneragent-0.1 7412) <stderr> ERROR:   File "C:\User
s\user\git\volttron\env\lib\ntpath.py", line 488, in abspath

2015-06-17 22:45:12,181 (listeneragent-0.1 7412) <stderr> ERROR:     path = _get
fullpathname(path)

2015-06-17 22:45:12,181 (listeneragent-0.1 7412) <stderr> ERROR: TypeError: must
 be (buffer overflow), not str

jhaack, 2015-06-18:
This issue is lowest priority and will not be worked on again until the next lengthy layover.

hashstat, 2015-06-22:
I got to thinking about this some more. I haven't tried running Volttron under [Cygwin](http://cygwin.org/), but I can't think of any reason it shouldn't run unmodified. Perhaps this is the path to recommend for those wishing to develop directly on Windows.

jhaack, 2018-05-30:
Windows 2010 can support a Docker container version of VOLTTRON

closed 2018-05-30 by jhaack