VOLTTRON/volttron #108: Logging can cause errors or unresponsive services

hashstat, 2015-06-24:
Logging handlers are currently created in the main thread using file objects that are not gevent-aware and log records can be emitted from both gevent and non-gevent contexts/threads. This can cause gevent threadlets to become unresponsive if a write to the log blocks. Wrapping the log file object in a gevent.fileobject.FileObject will fix the problem for gevent threadlets but will raise an exception in other threads if writing would block.

One possible solution is to write a smart handler or file object which is aware of the context and acts appropriately for that context.

craig8, 2016-10-26:
@kmonson want to take this one on/determine if necessary?

kmonson, 2016-10-26:
sure

kmonson, 2016-10-26:
Have we every actually observed this behavior?

acedrew, 2017-05-24:
@kmonson
Yes, we can definitely say we've observed this behavior now 😁 
Love the fact there was an existing issue for something that took me a few days of effort to debug... 🤕

jhaack, 2017-05-24:
Can you give steps for replicating?

acedrew, 2017-06-06:
Yes, point your logs at an rsyslog daemon, then kill rsyslog (may take some work to make it stay dead)
Where this was showing up in production was with verbose (DEBUG) logging turned on, and rsyslog shipping logs over the network. Slow connectivity may have exacerbated it, eventually rsyslog buffers would fill and it would begin to affect VOLTTRON

jhaack, 2018-05-30:
@kmonson  take a look at this in the current context