VOLTTRON/volttron #115: ZermoMQ generated identities contain non-Unicode characters/bytes

hashstat, 2015-07-02:
If a the ZMQ_IDENTITY socket option is not set on a socket before it connects, the endpoint will automatically generate a 5 byte identity prefixed with a NULL byte, with the remaining four bytes converted from a 4-byte integer. These values are not necessarily valid when encoded as Unicode, which is important because we serialize identities as JSON.

What is needed is to restrict identities to valid Unicode or to encode the automatically generated identities. I would recommend the former and have the VIP router drop messages (with a warning) sent by endpoints without valid identities so that we don't have to worry about encoding and decoding identities in so much code. The VIP *socket* module could automatically generate valid identities. This would become another requirement in the VIP specification.

Any thoughts or problems with this plan?

craig8, 2015-07-02:
+1 Though I didn't see any zmq documentation on how to restrict it.  I assume we would have to do that via some method?

hashstat, 2015-07-02:
@craig8 The router code would check for the NULL byte in identities on incoming messages and, instead of forwarding them, would generate a warning or error in the log. That would take care of the auto-generated identities. Our VIP socket module (volttron.platform.vip.socket) is where we would add the code to automatically generate an identity if one was not set and verify that the identity is Unicode-safe.

closed 2015-10-16 by hashstat