VOLTTRON/volttron #118: New pubsub subsystem may break compatibility with 2.x agents

hashstat, 2015-07-15:
The new pubsub subsystem is implemented over the RPC subsystem. As a consequence, the header and message is automatically serialized as JSON when sent and automatically deserialized at the remote end, so explicit serialization is not required as long as the values can be converted to valid JSON. 2.x agents, however, expect to send ZeroMQ frames and must explicitly serialize and deserialize messages. So new agents must also be explicit when communicating in a mixed environment or 2.x agents might not be able to interpret their messages.

hashstat, 2015-07-15:
As an example, send a pub/sub message using a new agent while running the listener agent.

fstshrk, 2015-07-16:
This is a real problem in a distributed system when two platforms can not communicate unless they run the same version of code.

Unfortunate.

commit 89bf96e6 (2015-07-16): Detect Content-Type and serialize messages if required. Fixes #118.
closed 2015-08-04 by hashstat