VOLTTRON/volttron #119: New VIP subsystem to improve sending to multiple recipients

hashstat, 2015-07-15:
Currently, if an agent must send a message to N recipients, it must iterate over the recipient list and send the message N times with only the recipient differing. Perhaps we should consider creating a subsystem which runs on the router that takes a list of recipients as well and send the message on behalf of the sender. This would help reduce the backlog of messages the router would need to handle.

hashstat, 2015-10-16:
Consider adding a new subsystem named 'manycast' or 'multicast' which expects a list of recipients, followed by an empty frame to mark the end of the list, followed by a subsystem and its associated frames.

Here is an example VIP message received by the router from the pubsub service requesting that the router distribute a pub/sub message on behalf of dan to the agents with identities alice, bob, and chuck:

sender | recipient | proto | user-id | msg-id | subsystem | peer1 | peer2 | peer3 | delim | subsystem | data
------ | --------- | ----- | ------- | ------ | --------- | ----- | ----- | ----- | ----- | --------- | ----
"pubsub" | "" | "VIP1" | "" | "" | "manycast" | "alice" | "bob" | "chuck" | "" | "rpc" | '{"jsonrpc"="2.0", "method"="pubsub.push", "params"=["dan", "", "some/topic", {}, "hello"]}'

The router would then turn this into the following outgoing messages:

recipient | sender | proto | user-id | msg-id | subsystem | data
--------- | ------ | ----- | ------- | ------ | --------- | ----
"alice" | "pubsub" | "VIP1" | "" | "" | "rpc" | '{"jsonrpc"="2.0", "method"="pubsub.push", "params"=["dan", "", "some/topic", {}, "hello"]}'
"bob" | "pubsub" | "VIP1" | "" | "" | "rpc" | '{"jsonrpc"="2.0", "method"="pubsub.push", "params"=["dan", "", "some/topic", {}, "hello"]}'
"chuck" | "pubsub" | "VIP1" | "" | "" | "rpc" | '{"jsonrpc"="2.0", "method"="pubsub.push", "params"=["dan", "", "some/topic", {}, "hello"]}'

Please note that these are notification messages and do not require a reply. The user-id is empty because the pubsub service communicates over an inproc: socket and is, therefore, part of the router process.

craig8, 2016-10-26:
With the modifications that @shwethanidd  is working on I think this will get taken care of in a different manner, but this is going to be addressed. @shwethanidd  please look this over and lets talk about it.

shwethanidd, 2016-10-31:
Since the platform pubsub is going to be part of router, the router will have list of subscriptions and it will send the message to all subscribers on the behalf of the publisher. 

However, if the message needs to be sent to only a subset of subscribers, then this feature will need to be added. List of peers needs to be part of the message similar to what is described above. The peer list needs to be part of the message similar to what is described above. Router will then only send to this peer list.

jhaack, 2018-05-30:
Completed

closed 2018-05-30 by jhaack