VOLTTRON/volttron #132: Refactoring "Agents" directory

jhaack, 2015-07-29:
Currently all agents live under the Agents directory off the base of the project. This can cause confusion as to what agents are essential to the platform, what are applications, and what are meant as examples. A new structure should be selected.

Proposal:

volttron/Agents goes away and is replaced with

volttron/
             services/    -> Agents which provide services to other agents.
                           core/  -> The core set of services for running the platform
                           contrib/  -> Services contributed back to the community, may be integrated into core

            applications/
                               examples/ -> Simple applications meant to illustrate a single feature
                               <organization name> -> Applications contributed by an organization
                               contrib/      -> General directory for other agents contributed back

The applications directory could potentially be a separate repository.

ahjortland, 2015-07-30:
I think this is a good idea - previously I have been confused about the origins and purposes of some agents.

Along these lines, would it also make sense to require/recommend a README in the base directory of each agent describing the origin, purpose, some implementation details, how it works, additional package requirement, etc.?  While some of this information can probably be inferred from the source, it might be more convenient to put important info in a central place.

jhaack, 2015-07-30:
A README would be very good. Some agents have content in the wiki but having it along with the agent makes sense too. In that README should be what version of the platform the agent has been verified with. This would clarify cases where agents were contributed back in a previous version but have not been checked by the authors against a newer version.

closed 2015-08-25 by craig8