VOLTTRON/volttron #133: ProtocolError: previous send operation is not complete

kmonson, 2015-07-30:
In the actuator agent I subscribe to 3 different topics. Occationally during startup I get these errors:

```
Traceback (most recent call last):
  File "/home/kyle/workspaces/volttron/env/local/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
    result = self._run(*self.args, **self.kwargs)
  File "/home/kyle/workspaces/volttron/volttron/platform/vip/agent/subsystems/pubsub.py", line 280, in subscribe
    self.rpc().call(peer, 'pubsub.subscribe', prefix, bus=bus).get()
  File "/home/kyle/workspaces/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 282, in call
    self.core().socket.send_vip(peer, 'RPC', [request], msg_id=ident)
  File "/home/kyle/workspaces/volttron/volttron/platform/vip/socket.py", line 225, in send_vip
    raise ProtocolError('previous send operation is not complete')
ProtocolError: previous send operation is not complete
<Greenlet at 0xb2fc002cL: subscribe(<volttron.platform.vip.agent.subsystems.pubsub.Pub, peer='pubsub', callback=<bound method ActuatorAgent.handle_set of <__main_, prefix=Topic(u'devices/actuators/set'))> failed with ProtocolError

Traceback (most recent call last):
  File "/home/kyle/workspaces/volttron/env/local/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
    result = self._run(*self.args, **self.kwargs)
  File "/home/kyle/workspaces/volttron/volttron/platform/vip/agent/subsystems/pubsub.py", line 280, in subscribe
    self.rpc().call(peer, 'pubsub.subscribe', prefix, bus=bus).get()
  File "/home/kyle/workspaces/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 282, in call
    self.core().socket.send_vip(peer, 'RPC', [request], msg_id=ident)
  File "/home/kyle/workspaces/volttron/volttron/platform/vip/socket.py", line 225, in send_vip
    raise ProtocolError('previous send operation is not complete')
ProtocolError: previous send operation is not complete
<Greenlet at 0xb2fc007cL: subscribe(<volttron.platform.vip.agent.subsystems.pubsub.Pub, peer='pubsub', callback=<bound method ActuatorAgent.handle_schedule_reques, prefix=Topic(u'devices/actuators/schedule/request'))> failed with ProtocolError
```

I assume that the second and third subscription did not take in this case.

kmonson, 2015-07-30:
It seems more likely to occur if I setup the subscription as decorators.

hashstat, 2015-07-30:
I think this is an unintended consequence of using the ZMQ_IMMEDIATE option on the socket. I'll take a look at it.

kmonson, 2015-07-30:
I'll check in the current version of the actuator agent into the driver
branch if you want to test with that.

hashstat, 2015-07-30:
It's not necessary. I am able to easily reproduce.

> I'll check in the current version of the actuator agent into the driver
> branch if you want to test with that.

hashstat, 2015-07-30:
The problem is that subscriptions are being attempted before the VIP socket is fully connected and the **IMMEDIATE** option is used causing send operations to block until the connection completes. **IMMEDIATE** is used to prevent filling the buffer with messages that may overwhelm the router once the socket connects and to provide immediate feedback about the availability of the router. In this case, the first call to *subscribe()* is blocking, waiting for the connection to the router to finish, while other tasks are also attempting to subscribe. Because the first task is blocking and the socket state indicates that a partial send is in progress, the exception is raised and will be for all subsequent subscriptions until the socket finishes connecting.

The proposed solution is to add **NOBLOCK** to the default flags for send operations which would cause a *zmq.Again* exception to be raised instead of blocking when a message cannot be sent immediately. A sender can either check the socket's *events* attribute or call *poll()* to determine if the socket is ready for sending. This is probably closer to the behavior one would expect from an asynchronous network application.

The proposed fix will have consequences. The socket will be much more likely to raise an exception on send, potentially causing agents to fail in ways that have rarely been seen before. Ideally, any code which exercises a ZMQ socket should already be wrapped in a try-except block and take appropriate actions on socket errors rather than let exceptions propagate, potentially killing the process. That needs to be the case everywhere. We could provide a context manager which automatically handles catching the *zmq.Again* exception and blocks until a resend can occur (possibly with options for the number of retries and timeout period) to ease handling this common case.

I would like to use this thread to discuss the proposed solution and solicit alternate approaches.

kmonson, 2015-07-30:
If it's supposed to block until the socket is connected then why am I
seeing the problem when I don't use a decorator and instead do it like this:

self.vip.pubsub.subscribe(peer='pubsub',
                                      prefix=topics.ACTUATOR_GET(),
                                      callback=self.handle_get)

self.vip.pubsub.subscribe(peer='pubsub',
                                      prefix=topics.ACTUATOR_SET(),
                                      callback=self.handle_set)

self.vip.pubsub.subscribe(peer='pubsub',

prefix=topics.ACTUATOR_SCHEDULE_REQUEST(),
                                      callback=self.handle_schedule_request)

I would assume that the first call would block until it was connected
before moving on.

Could we assure that the socket was connected before getting to any point
where a send might occur?

hashstat, 2015-07-30:
> If it's supposed to block until the socket is connected then why am I
> seeing the problem when I don't use a decorator and instead do it like this:

The *subscribe()* method returns a greenlet which was spawned to perform the RPC call and the finishing work. Somewhere down in that call chain *send_vip_object()* is called and is blocking.

> I would assume that the first call would block until it was connected
> before moving on.

Because *subscribe()* spawns a greenlet, it returns immediately. You need to call their *get()* methods to serialize them.

```python
self.vip.pubsub.subscribe(
    peer='pubsub', prefix=topics.ACTUATOR_GET(), callback=self.handle_get).get()

self.vip.pubsub.subscribe(
    peer='pubsub', prefix=topics.ACTUATOR_SET(), callback=self.handle_set).get()

self.vip.pubsub.subscribe(
    peer='pubsub', prefix=topics.ACTUATOR_SCHEDULE_REQUEST(),
    callback=self.handle_schedule_request).get()
```

> Could we assure that the socket was connected before getting to any point
> where a send might occur?

That is a good suggestion. We could automatically call *poll()* at the beginning of the function to block before all the state checking. Or we could wrap the send methods in a lock so that one send operation must complete before another, but I would rather not use that approach.

kmonson, 2015-07-30:
I must be looking at the wrong subscribe function. I don't see where it
returns a greenlet.

I tried that and it worked. I'll mess around and see if that fixes the
problem for me at least.

hashstat, 2015-07-30:
> I must be looking at the wrong subscribe function. I don't see where it returns a greenlet.

Sorry, it is returned by the magical *spawn* decorator:

```python
@spawn
def subscribe(self, peer, prefix, callback, bus=''):
```

The decorator wraps the method up in a greenlet.

kmonson, 2015-07-30:
Ah, ok.

kmonson, 2015-07-31:
Looks fixed to me.

commit 7fb6fb3c (2015-07-30): Block at top of VIP socket send methods. Fixes #133.
commit c75fb4b8 (2015-08-03): Modified Socket class to use both locking and polling. Fixes #133.

This modification reduces the number of calls to poll() by using a
recursive lock to protect access to the send operations. poll() is only
called for non-blocking sends.
closed 2015-08-04 by hashstat