VOLTTRON/volttron #146: agent shutdown

craig8, 2015-08-06:
when running volttron-ctl shutdown the indivdual agents are not shutting down properly.

hashstat, 2015-08-10:
The 2.x base agent subscribed to the '/platform/shutdown' topic and would use it to do clean up and quit. The 3.x base agent no longer subscribes to that topic, which is why agents stopped responding to it. Now that agents have the onstop and onfinish signals for cleanup, is it necessary to continue sending the shutdown topic? If not, we can cleanup the shutdown method to only include @craig8's changes.

craig8, 2015-08-10:
I don't think it is.   The /platform topic is however subscribed to by the SQLHistorian as a way of determining when a PlatformAgent comes and goes.    No need for the root of the agents to subscribe to it however.

PR #147 (2015-08-06): Added a fix that will stop the agents before doing the shutdown part …
Fixes #146
closed 2015-08-25 by craig8