VOLTTRON/volttron #159: Investigate intermittent communication issue

jhaack, 2015-08-25:
Shows up in ForwardingHistorian attempting to publish data to a remote historian. At some point, the connection gets lost and hangs on the publish attempt.

Investigating to find root cause.

hashstat, 2015-08-31:
It appears the root cause has been discovered and is related to [libzmq issue 1453](/zeromq/libzmq/issues/1453). File descriptors for TCP connections which die unnaturally do not get detected as such and do not get cleaned up by ZMQ ROUTER sockets. Because the dead socket holds the identity of the remote endpioint, it essentially locks out the peer from future communications until the ROUTER socket is unbound and rebound or the application is restarted.

This is a high priority issue that will likely require changes to libzmq.

hashstat, 2015-08-31:
Using TCP keepalive may be an option here. ZMQ supports it, but it defaults to the OS defaults, which is not to not enable keepalive. I will experiment with enabling TCP keepalive on the VIP sockets and see what it gets us.

hashstat, 2015-08-31:
A [helpful link](http://tldp.org/HOWTO/TCP-Keepalive-HOWTO/overview.html) for those who want to know more about TCP keepalive.

hashstat, 2015-08-31:
Initial testing of TCP keepalive using a minimal test case showed success. After missing packets for the expected maximum period, the underlying socket was disconnected and the remote endpoint reconnected automatically.

The next question is that of keepalive settings. There are 3 values to worry about: idle time, retry interval, and retry count. Currently, I've set them to 600, 60, and 10 respectively, meaning that after 10 minutes (600 seconds) of inactivity a keepalive probe will be sent every 60 seconds up to a maximum of 10 times or until a reply is received. This means it could take up to 20 minutes to detect a dead socket. Does that seem reasonable or is it too long? It is definitely much shorter than the Linux defaults of 7200, 75, and 9 (total of ~2h11m) but possibly not short enough since the identity cannot be reused until the socket is disconnected. My test case used 60, 20, and 6 which gave around to 3 minutes to detect a dead peer.

What values do others think reasonable?

craig8, 2015-09-01:
I think 3-5 minutes would be a good default and then allow the user to customize to meet their specific needs.

hashstat, 2015-09-02:
The team supports a 3 minute idle time followed by 6 probes every 20 seconds.

hashstat, 2015-09-02:
Fixed by commit 82c3957377796fdb9a9a9356dd059c4002994045.

craig8, 2015-09-03:
It looks like this didn't fix so reopening.

closed 2015-09-24 by craig8