VOLTTRON/volttron #166: socket for publish/subscription does not exist

sumanth13, 2015-08-27:
Hi ,
I am new to volttron. I tried to build test agent in the way it is shown in the website but i am getting some errors saying publish socket and subscription socket does not exist . Can some help me please?

hashstat, 2015-08-31:
Please tell us what branch you are on and the URL of the website you are referring to. We are currently updating documentation for the 3.x release, so some of the instructions for 2.x require some modification.

Most likely, all you need to do is add the strudel (@) to the root of the socket paths. For instance,
*ipc://home/user/.volttron/run/publish* would become *ipc://@/home/user/.volttron/run/publish*. This is because with the new version, the sockets are abstract by default.

sumanth13, 2015-09-02:
I am currently working on agent development and I followed the following link to develop the sample agent. 
Weblink: https://github.com/VOLTTRON/volttron/wiki/AgentDevelopment

hashstat, 2015-09-02:
It sounds like you are following instructions for 2.x but checked out either the master or 3.x branch. There have been changes that will cause some of the instructions for 2.x to fail on the newer branches. Please try `git branch` to determine which branch you are on and let us know. It will help us know how to direct you next. Thanks.

sumanth13, 2015-09-07:
Hi,
I tried git branch and it says 2.x.

*Thanks & Regards,*
*Nallabolu Sumanth Reddy*
*Electrical and Computer Engineering,*
*Michigan Technological University,*
*Houghton,MI-49931.*

bruskauff, 2015-09-11:
Hello,

I'm having a similar issue at the moment. I'm following instructions to set up Eclipse for development in VOLTTRON 2.0: User Guide from November 2014. I'm in section 4.5.3.2 (Configure a Run Configuration for the Listener Agent)
I tried adding the strudel with no luck.
"git branch" yields me "* 2.x"

The full error message is below. Thanks for any help!

Full error message:
warning: subscription socket does not exist: /home/parallels/.volttron/run/subscribe
warning: publish socket does not exist: /home/parallels/.volttron/run/publish
{"relativeCreated":66.09606742858887,"process":2923,"args":[],"module":"agent","funcName":"setup","exc_text":null,"name":"__main__","thread":140235744671552,"created":1441992935.465094,"threadName":"MainThread","msecs":465.09408950805664,"filename":"agent.py","levelno":20,"processName":"MainProcess","pathname":"/home/parallels/volttron/Agents/ListenerAgent/listener/agent.py","lineno":83,"msg":"hello","levelname":"INFO"}

hashstat, 2015-09-16:
Sorry for the delayed response. I have been on vacation.

Thanks for the additional information. I checked into the 2.x code and do not see anything anomalous. The subscribe and publish sockets are being created in the usual locations and using the strudel is unnecessary in this branch.

The platform (*env/bin/volttron*) must be running for the sockets to exist. Ensure that you are running the *Platform Run Configuration* created in section 4.3.5.1 before executing any agent run configurations. Executing `netstat -xl | grep volttron` from a terminal should list at least three sockets, which include the control, publish, and subscribe sockets. If that command does not display these three sockets, the platform is not running and must be started before any agents.

Let us know how it goes.

sumanth13, 2015-09-16:
*Platform Run Configuration* created in section 4.3.5.1 is to run the
volttron within eclipse and I am not using eclipse. Even the command (netstat
-xl | grep volttron) is not showing any sockets lists.

*Thanks & Regards,*
*Nallabolu Sumanth Reddy*
*Electrical and Computer Engineering,*
*Michigan Technological University,*
*Houghton,MI-49931.*

hashstat, 2015-09-16:
@sumanth13, you need to start volttron before any agents. I also do not use eclipse and dedicate a terminal window for running volttron. You can just execute `volttron` if you activated the environment with `. env/bin/activate`. Otherwise, you will need to execute with the path, `env/bin/volttron`, assuming the current working directory is the volttron directory.

sumanth13, 2015-09-16:
Yes I did that too

sumanth13, 2015-09-16:
Can you please give me your suggestion on my new post too.

hashstat, 2015-09-16:
@sumanth13, if `netstat -xl | grep volttron` is not listing any sockets, then volttron is likely not running or perhaps there are permission issues. But permission issues should cause volttron to display an error and halt. Please also try `pgrep -a volttron`. Again, if you get nothing, volttron is not running.

Another issue others have run into which causes weird failures is when some of the installation or configuration steps were performed as root or when volttron is run as root (perhaps using sudo). Please check that root does not own any files in the volttron directories: `find . ~/.volttron -user root`. Again, run this command from the directory where you cloned the volttron repo. If *find* lists any files, weirdness will ensue. You can add an *-execdir* argument to the *find* command to chown the files.

hashstat, 2015-09-22:
@sumanth13, @bruskauff: Is this issue resolved? Were you able to get things going? We don't want to leave you hanging.

bruskauff, 2015-09-22:
Whoops, forgot to respond here, sorry about that! Eclipse has been giving me other issues that I haven't had the patience to deal with so I'm not going to bother dealing with it. Sorry I can't confirm if it was a program issue or user issue, but I think it's safe to say I messed up a configuration point.

Aside from Eclipse VOLTTRON is publishing fine between a few of the agents I have running. Thanks for all your hard work :)

sumanth13, 2015-09-23:
Same with me. I just left it for a while. Now I am working on openadr. Thanks a lot for the support.

bruskauff, 2015-09-25:
I would like to add a additional note for anyone who has this issue in the future. @hashstat, this may be something you would be interested in as well. I just had the nonexistent socket error happen again after I accidentally typed "volttron -l volttron.log -vv&" for the second time into the volttron instance running. I restarted volttron and it worked fine. Hope this helps some :)

sumanth13, 2015-09-27:
What is the basic communication protocol in volttron 3.x? like internet or
TCP/IP or something else

*Thanks & Regards,*
*Nallabolu Sumanth Reddy*
*Electrical and Computer Engineering,*
*Michigan Technological University,*
*Houghton,MI-49931.*

hashstat, 2015-09-28:
> What is the basic communication protocol in volttron 3.x? like internet or TCP/IP or something else

@sumanth13 also asked this question via email and it will be answered there. I am closing this thread.

closed 2015-09-28 by hashstat