VOLTTRON/volttron #167: Periodic methods calls cease if the method raises an unhandled exception.

kmonson, 2015-08-28:


hashstat, 2015-08-31:
I don't think this should be considered a bug. The unhandled exception is likely to occur again in a subsequent run of the periodic. If that is the desired behavior, the agent developer should catch and log the exception.

I would actually recommend that unhandled exceptions in periodics be logged and then cause the agent to halt. But again, this should probably be explicitly handled by the agent developer. We want to be careful not to introduce too many automatic helps that may mask true errors or make them difficult to respond to.

Thoughts?

kmonson, 2015-09-01:
I was bit by this when the driver's periodic scrape was failing and raising
an  exception that I did not catch. The current behavior kept the driver
from recovering when communication with the device was restored. I've now
handled it, but it was surprising way for the framework to act.

For me, the expected behavior is for the periodic to continue to be called
even if it fails. This is the behavior of frameworks like twistd and
tornado.

hashstat, 2015-09-02:
After discussing the issue with the team, we agreed that wrapping the periodic call in a try-except block and logging the exceptions is probably the best approach.

commit e93b2add (2015-09-02): Log exceptions in periodics and let them keep running. Fixes #167.
closed 2015-09-02 by hashstat