VOLTTRON/volttron #168: Logging exceptions in agents causes JSON serialization error

hashstat, 2015-08-31:
The *JSONFormatter* logging formatter installed by *volttron.platform.agent.utils:setup_logging* serialized logger output as JSON to be passed via stderr to the platform for appending to the actual VOLTTRON log. However, *Exception* objects are not serializeable as JSON and an exception is raised. Here is an example of the error from the log file:

```
2015-08-28 08:38:59,118 (forwarderagent-3.0 6934) volttron.platform.agent.base_historian ERROR: An unhandled exception has occured while publishing to historian.
2015-08-28 08:38:59,119 (forwarderagent-3.0 6934) <stderr> ERROR: Traceback (most recent call last):
2015-08-28 08:38:59,119 (forwarderagent-3.0 6934) <stderr> ERROR:   File "/usr/lib/python2.7/logging/__init__.py", line 851, in emit
2015-08-28 08:38:59,120 (forwarderagent-3.0 6934) <stderr> ERROR:     msg = self.format(record)
2015-08-28 08:38:59,120 (forwarderagent-3.0 6934) <stderr> ERROR:   File "/usr/lib/python2.7/logging/__init__.py", line 724, in format
2015-08-28 08:38:59,120 (forwarderagent-3.0 6934) <stderr> ERROR:     return fmt.format(record)
2015-08-28 08:38:59,120 (forwarderagent-3.0 6934) <stderr> ERROR:   File "/home/volttron/volttron-3x/volttron/platform/agent/utils.py", line 228, in format
2015-08-28 08:38:59,121 (forwarderagent-3.0 6934) <stderr> ERROR:     return jsonapi.dumps(dct)
2015-08-28 08:38:59,121 (forwarderagent-3.0 6934) <stderr> ERROR:   File "/home/volttron/volttron-3x/env/local/lib/python2.7/site-packages/zmq/utils/jsonapi.py", line 40, in dumps
2015-08-28 08:38:59,121 (forwarderagent-3.0 6934) <stderr> ERROR:     s = jsonmod.dumps(o, **kwargs)
2015-08-28 08:38:59,122 (forwarderagent-3.0 6934) <stderr> ERROR:   File "/home/volttron/volttron-3x/env/local/lib/python2.7/site-packages/simplejson/__init__.py", line 397, in dumps
2015-08-28 08:38:59,122 (forwarderagent-3.0 6934) <stderr> ERROR:     **kw).encode(obj)
2015-08-28 08:38:59,122 (forwarderagent-3.0 6934) <stderr> ERROR:   File "/home/volttron/volttron-3x/env/local/lib/python2.7/site-packages/simplejson/encoder.py", line 275, in encode
2015-08-28 08:38:59,122 (forwarderagent-3.0 6934) <stderr> ERROR:     chunks = self.iterencode(o, _one_shot=True)
2015-08-28 08:38:59,123 (forwarderagent-3.0 6934) <stderr> ERROR:   File "/home/volttron/volttron-3x/env/local/lib/python2.7/site-packages/simplejson/encoder.py", line 357, in iterencode
2015-08-28 08:38:59,123 (forwarderagent-3.0 6934) <stderr> ERROR:     return _iterencode(o, 0)
2015-08-28 08:38:59,123 (forwarderagent-3.0 6934) <stderr> ERROR:   File "/home/volttron/volttron-3x/env/local/lib/python2.7/site-packages/simplejson/encoder.py", line 252, in default
2015-08-28 08:38:59,124 (forwarderagent-3.0 6934) <stderr> ERROR:     raise TypeError(repr(o) + " is not JSON serializable")
2015-08-28 08:38:59,124 (forwarderagent-3.0 6934) <stderr> ERROR: TypeError: ProtocolError('previous send operation is not complete',) is not JSON serializable
2015-08-28 08:38:59,124 (forwarderagent-3.0 6934) <stderr> ERROR: Logged from file base_historian.py, line 370
```

The offending code which triggered the exception is this exception handling code, specifically the final line:

```python
try:
    self.publish_to_historian(to_publish_list)
except Exception as exp:
    _log.error("An unhandled exception has occured while " \
               "publishing to historian.")
    _log.exception(exp)
```

craig8, 2015-08-31:
@hashstat I believe this code is also in the other historians.  What would you think would be the best way to fix this?  In the agent, not logging the exception or creating a formatter that would handle exceptions?  Or a third option?

hashstat, 2015-08-31:
@craig8 I think modifying the formatter to handle exceptions is probably the right way to go.

hashstat, 2015-09-02:
Looks like this is an error using the exception log facility. The code should actually read like this:

```python
try:
    self.publish_to_historian(to_publish_list)
except Exception:
    _log.exception("An unhandled exception has occured while " \
               "publishing to historian.")
```

There is no need to pass the exception to the log method. In fact, that is the cause of the error because it isn't JSON serializable. The exception logger will automatically handle formatting the current exception.

hashstat, 2015-09-02:
This is also happening in *services/core/MasterDriverAgent/master_driver/driver.py* and should be fixed there as well.