VOLTTRON/volttron #184: Profiling VOLTTRON process

jhaack, 2015-09-17:
Use Python profiling to find resource usage of VOLTTRON subsystems

craig8, 2015-10-02:
@jhaack are we sure this is part of 3.0 or is it already done?

jhaack, 2015-10-02:
Changed to 3.1, we still need to investigate this.

jhaack, 2016-10-10:
Produce a readthedocs page on results and procedure for repeating it yourself.

craig8, 2016-10-26:
@mikeroup did you provide a page on readthedocs about your profileing results?

mikeroup, 2016-10-26:
@craig8 I haven't gotten to it yet

acedrew, 2019-10-10:
@tnesztler shared some memory profiles in Slack that were pretty interesting. What is the actual goal of this? I feel like it's good to do, but I know there have been a lot of performance and reliability improvements since this issue.
I also think there's a huge difference in what can be tested with virtual devices and the like and what a real-world environment looks like. From my experience, I/O and remote devices affect performance a lot more than anything else.

tnesztler, 2019-10-10:
Relating to I/O, our main issues with a large installation of Volttron over the past 8 months (1.8M points each 10min over 4 instances on 2 PCs) have been memory usage (an instance would eat ~7-8GB) and network latency (having to talk to equipment located +1000 miles away). We have to use Failover agents with each agent (historian, bacnet proxy and master driver) to prevent any downtime since the system could kill an agent due to high mem. usage.
I would say that Volttron has been pretty awesome nonetheless compared to other products limiting the number of trends to something like 2000 points.