VOLTTRON/volttron #189: Unable to run the script file (volttron/scripts/management-service-demo/run-demo)

sumanth13, 2015-09-30:
Whenever I try to run this scrpit then a terminal pops up and closes automatically with any display. Can someone help me with this issue?

hashstat, 2015-09-30:
I would be happy to help. I haven't run that script before, so I can't give a flippant "it works on my system" answer*. At least not yet. :wink:

<sub>* This comment is an attempt at humor and in no way should be taken to indicate that anyone on the VOLTTRON team would actually provide a flippant answer.</sub>

hashstat, 2015-09-30:
@sumanth13, can you provide some more detail about how you run the script? Is there a guide/wiki page you are following? What is the exact command you execute? Are you doing that in a terminal with VOLTTRON activated?

Let me know the steps to repeat so that I can provide you quick and accurate feedback. Thanks.

sumanth13, 2015-09-30:
When I searched how to run a .sh file I found two ways
1) By making the file executable and then double clicking on it.
2) Running the file in terminal i.e., sh <filename>.sh

The above mentioned issue is when I tried to first way. 
When I tried the second way by activating volttron it says: Please execute from root of volttron repository.

hashstat, 2015-09-30:
Yeah. The problem with (1) is that it launches the terminal and, unless the terminal is configured otherwise, it closes after execution providing no opportunity to examine the output. The script is designed to be run from an activated shell. Open a terminal and do the following:

```
~$ cd $VOLTTRON_HOME
volttron$ . env/bin/activate
(volttron) volttron$ bash scripts/management-service-demo/run-demo.sh
```

Replace $VOLTTRON_HOME with the path where you checked out the VOLTTRON repository.

sumanth13, 2015-09-30:
Now I am able to run the file and launch the browser but there is a problem
is loading the page. I am sending the screen shot of the browser.

​

*Thanks & Regards,*
*Nallabolu Sumanth Reddy*
*Electrical and Computer Engineering,*
*Michigan Technological University,*
*Houghton,MI-49931.*

craig8, 2015-09-30:
ok so from the terminal can you exectute the comman
    VOLTTRON_HOME=/tmp/v1home volttron-ctl status

This should show the status of the agents on the 1st volttron instance.  There are 3 instances v1home, v2home, and v3home created when you run the demo script.  You should see agents in a running state.

hashstat, 2015-09-30:
> Now I am able to run the file and launch the browser but there is a problem is loading the page. I am sending the screen shot of the browser.

@sumanth13, it looks like you replied via an email client and the screenshot did not make it through. @craig8 believes the problem your are now experiencing might be due to other instances of volttron running. If you don't mind killing them, try `pkill volttron` and run the script again. If this doesn't help, please attach your screenshot using Github's issues interface rather than replying via your email client. Thanks.

sumanth13, 2015-10-01:
When I tried the above command (volttron-ctl status) it waits for some 20-30 seconds and then displays: status: operation timed out. I tried pkill volttron also but the result is same. I am sending screenshots
[image]
[image]
[image]
[image]

 so that you can understand the issue more clearly

sumanth13, 2015-10-01:
One of my friends is also trying the same thing. He found that the script file is running well in linux mint but was unable to do it in ubuntu12.04.

craig8, 2015-10-01:
In order to use the volttron instances that are created with the demo script you must prefix it with the VOLTTRON_HOME variable.

    VOLTTRON_HOME=/tmp/v1home volttron-ctl status

Note that this is not the same thing as

    volttron-ctl status

which is what I see you have entered in your shell.  The one you entered is attempting to contact a default instance of volttron with the VOLTTRON_HOME set to $HOME/.volttron.  As you are running the demo script that instance is not available.

In otherwords, you will use volttron-ctl without a VOLTTRON_HOME when you are running a default volttron e.g. if you started volttron like

    volttron -vv -l volttron.log& 

then you could do a 

    volttron-ctl status

and you should receive a response.

I hope this helps.

craig8, 2015-10-01:
One other thing.  In your images I don't see volttroncentral agent as one of the ones that was started.  I assume this is because it is further up on the command window buffer.  The volttroncentral agent is what is responsible for starting the webserver.

sumanth13, 2015-10-01:
[image]

sumanth13, 2015-10-01:
Above is what I got when I run check the status

craig8, 2015-10-01:
Awesome...so that means that the vc errored out or something happened.  You can try restarting it with the command

    VOLTTRON_HOME/tmp/v1home volttron-ctl start --tag vc

if after running the above status again after that then please upload the log file located at /tmp/v1home/volttron.log

From that I should be able to see what is going on (hopefully).  Also, I don't see in this thread which branch you have checked out from.  3.x or master?

Thanks

sumanth13, 2015-10-01:
2015-09-30 22:20:35,778 () volttron.platform.main DEBUG: open file resource limit increased from 1024 to 4096
2015-09-30 22:20:35,778 () volttron.platform.main WARNING: insecure mode on directory: /tmp/v1home
2015-09-30 22:20:35,779 () volttron.platform.main DEBUG: using key file /tmp/v1home/curve.key
2015-09-30 22:20:35,779 () volttron.platform.main WARNING: insecure mode on key file
2015-09-30 22:20:35,779 () volttron.platform.main WARNING: empty key file; VIP encryption is disabled!
2015-09-30 22:20:35,784 () volttron.platform.main ERROR: Unhandled exception in router loop
Traceback (most recent call last):
  File "/home/sumanth/Desktop/V3/volttron-3.x/volttron/platform/main.py", line 573, in router
    monitor=opts.monitor).run()
  File "/home/sumanth/Desktop/V3/volttron-3.x/volttron/platform/main.py", line 271, in run
    self.start()
  File "/home/sumanth/Desktop/V3/volttron-3.x/volttron/platform/vip/router.py", line 120, in start
    self.setup()
  File "/home/sumanth/Desktop/V3/volttron-3.x/volttron/platform/main.py", line 257, in setup
    sock.bind(self.local_address)
  File "zmq/backend/cython/socket.pyx", line 489, in zmq.backend.cython.socket.Socket.bind (zmq/backend/cython/socket.c:4824)
  File "zmq/backend/cython/checkrc.pxd", line 25, in zmq.backend.cython.checkrc._check_rc (zmq/backend/cython/socket.c:7055)
    raise ZMQError(errno)
ZMQError: Address already in use
2015-09-30 22:20:37,050 () volttron.platform.auth INFO: 'vip' 'localhost:1002:1002:9111' 'NULL' []
2015-09-30 22:20:37,053 () volttron.platform.aip INFO: starting agent /tmp/v1home/agents/de49899b-362b-405c-8526-7d22c19495c1/volttroncentralagent-0.1
2015-09-30 22:20:37,058 () volttron.platform.aip INFO: agent /tmp/v1home/agents/de49899b-362b-405c-8526-7d22c19495c1/volttroncentralagent-0.1 has PID 9129
2015-09-30 22:20:37,264 (volttroncentralagent-0.1 9129) <stderr> ERROR: Traceback (most recent call last):
2015-09-30 22:20:37,264 (volttroncentralagent-0.1 9129) <stderr> ERROR:   File "<string>", line 1, in <module>
2015-09-30 22:20:37,264 (volttroncentralagent-0.1 9129) <stderr> ERROR:   File "/tmp/v1home/agents/de49899b-362b-405c-8526-7d22c19495c1/volttroncentralagent-0.1/volttroncentral/vc.py", line 78, in <module>
2015-09-30 22:20:37,264 (volttroncentralagent-0.1 9129) <stderr> ERROR:     from volttron.platform.async import AsyncCall
2015-09-30 22:20:37,264 (volttroncentralagent-0.1 9129) <stderr> ERROR:   File "/home/sumanth/Desktop/V3/volttron-3.x/volttron/platform/async.py", line 78, in <module>
2015-09-30 22:20:37,264 (volttroncentralagent-0.1 9129) <stderr> ERROR:     class Threadlet(threading.Thread):
2015-09-30 22:20:37,264 (volttroncentralagent-0.1 9129) <stderr> ERROR:   File "/home/sumanth/Desktop/V3/volttron-3.x/volttron/platform/async.py", line 111, in Threadlet
2015-09-30 22:20:37,264 (volttroncentralagent-0.1 9129) <stderr> ERROR:     __init__.__doc__ = threading.Thread.__init__.__doc__ + __init__.__doc__
2015-09-30 22:20:37,264 (volttroncentralagent-0.1 9129) <stderr> ERROR: TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
2015-09-30 22:20:38,476 () volttron.platform.auth INFO: 'vip' 'localhost:1002:1002:9210' 'NULL' []
2015-09-30 22:20:38,479 () volttron.platform.aip INFO: starting agent /tmp/v1home/agents/ca5e2b94-41b9-463c-8f39-67322082e7ef/platformagentagent-0.1
2015-09-30 22:20:38,484 () volttron.platform.aip INFO: agent /tmp/v1home/agents/ca5e2b94-41b9-463c-8f39-67322082e7ef/platformagentagent-0.1 has PID 9228
2015-09-30 22:20:38,701 (platformagentagent-0.1 9228) platformagent.agent INFO: my identity platform.agent address: ipc://@/tmp/v1home/run/vip.socket
2015-09-30 22:20:38,701 (platformagentagent-0.1 9228) platformagent.agent DEBUG: Exception 
2015-09-30 22:20:38,706 (platformagentagent-0.1 9228) platformagent.agent DEBUG: update_sibling_address_cache set([])
2015-09-30 22:20:38,707 () volttron.platform.auth INFO: 'vip' 'localhost:1002:1002:9228' 'NULL' []
2015-09-30 22:20:38,743 () volttron.platform.auth INFO: 'vip' 'localhost:1002:1002:9233' 'NULL' []
2015-09-30 22:20:41,710 (platformagentagent-0.1 9228) <stderr> ERROR: Traceback (most recent call last):
2015-09-30 22:20:41,710 (platformagentagent-0.1 9228) <stderr> ERROR:   File "/home/sumanth/Desktop/V3/volttron-3.x/env/local/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
2015-09-30 22:20:41,710 (platformagentagent-0.1 9228) <stderr> ERROR:     result = self._run(*self.args, **self.kwargs)
2015-09-30 22:20:41,710 (platformagentagent-0.1 9228) <stderr> ERROR:   File "/tmp/v1home/agents/ca5e2b94-41b9-463c-8f39-67322082e7ef/platformagentagent-0.1/platformagent/agent.py", line 384, in starting
2015-09-30 22:20:41,710 (platformagentagent-0.1 9228) <stderr> ERROR:     _, _, my_id = self.vip.hello().get(timeout=3)
2015-09-30 22:20:41,710 (platformagentagent-0.1 9228) <stderr> ERROR:   File "/home/sumanth/Desktop/V3/volttron-3.x/env/local/lib/python2.7/site-packages/gevent/event.py", line 233, in get
2015-09-30 22:20:41,711 (platformagentagent-0.1 9228) <stderr> ERROR:     result = self.hub.switch()
2015-09-30 22:20:41,711 (platformagentagent-0.1 9228) <stderr> ERROR:   File "/home/sumanth/Desktop/V3/volttron-3.x/env/local/lib/python2.7/site-packages/gevent/hub.py", line 338, in switch
2015-09-30 22:20:41,711 (platformagentagent-0.1 9228) <stderr> ERROR:     return greenlet.switch(self)
2015-09-30 22:20:41,711 (platformagentagent-0.1 9228) <stderr> ERROR: Timeout: 3 seconds
[... cut ...]
2015-09-30 22:25:43,723 (sqlhistorianagent-0.1 9576) volttron.platform.agent.base_historian DEBUG: Getting oldest outstanding to publish.
2015-09-30 22:25:43,723 (sqlhistorianagent-0.1 9576) volttron.platform.agent.base_historian DEBUG: Reading from/waiting for queue.
2015-09-30 22:25:45,211 (platformagentagent-0.1 7231) platformagent.agent DEBUG: update_sibling_address_cache set([])
2015-09-30 22:26:08,763 (platformagentagent-0.1 9228) platformagent.agent DEBUG: update_sibling_address_cache set([])
2015-09-30 22:26:15,212 (platformagentagent-0.1 7231) platformagent.agent DEBUG: update_sibling_address_cache set([])

sumanth13, 2015-10-01:
log file is not uploading so i have copy pasted the entire log file.

sumanth13, 2015-10-01:
it is 3.x

craig8, 2015-10-01:
Ahhh haaa...So I believe the root issue of your not being able to get this working is right here

File "/home/sumanth/Desktop/V3/volttron-3.x/volttron/platform/main.py", line 257, in setup
sock.bind(self.local_address)
File "zmq/backend/cython/socket.pyx", line 489, in zmq.backend.cython.socket.Socket.bind (zmq/backend/cython/socket.c:4824)
File "zmq/backend/cython/checkrc.pxd", line 25, in zmq.backend.cython.checkrc.check_rc (zmq/backend/cython/socket.c:7055)
raise ZMQError(errno)
ZMQError: Address already in use

Specifically this means that you have some rogue volttron processes that were running before you ran the script again.  When this happens to me I like taking the sledge hammer approach to cleaning things up.

   killall -9 python

if you execute that command 2x you should get a no processes found response.

Then you should be able to run the script successfully.

Please let me know if this works out for you.

craig8, 2015-10-02:
So I made up a vm with the latest version of ubuntu and was able to run it successfully with the 3.x branch.

[image]

Now we need to make sure that you are using the latest from the 3.x branch.  Please execute the following.

```
git log
```

Do you get the following output?

[image]

If not please 

```
git pull
killall -9 python
./scripts/management-service-demo/run-demo.sh 
```

sumanth13, 2015-10-03:
Thank you very much for the support. I have installed linux mint and now it is working. No issues for now.

craig8, 2015-10-08:
I am going to close this as it seems we have something that works for you now.

closed 2015-10-08 by craig8