VOLTTRON/volttron #190: shutdown --platform does not kill agents before shutting down the platform

craig8, 2015-09-30:
I know we addressed this before, but I couldn't find the issue and so I am creating this one.  I have 4 agents running on a platform and call volttron-ctl shutdown --platform.  The agents are still running after this process. using ps aux|grep python.

hashstat, 2015-10-01:
Issues #113 and #146 are possibly related.

Currently, only agents started by the platform and under its control are stopped when a shutdown is issued, by sending a SIGINT signal. The platform also sends a */platform/shutdown* pub/sub topic, but 3.x agents are not listening for it by default as 2.x agets do. I don't think it would be appropriate for remote agents to shutdown when the platform goes away or issues a shutdown.

@craig8, are your agents started by the platform or outside the platform, such as through pydev?

craig8, 2015-10-02:
@hashstat when I saw this happen the agents were started through the platform.  The problem that I see happened after using multiple times of my "config" scripts.  Which are using the core volttron-ctl commands to add/remove agents from the platform.  I have not had a chance to setup an example yet, I will do so in the AM and try and reproduce results.

craig8, 2015-10-02:
This seems to not be an error.  I must have had multiple volttron instances and didn't clean up the correct instance.  Therefore there would still be processes.

closed 2015-10-02 by craig8