VOLTTRON/volttron #194: Agenda for 10/16 office hours

jhaack, 2015-10-02:
Use this issue to gauge interest in topics for the next office hours

jhaack, 2015-10-02:
Detailed discussion/walkthrough of VOLTTRON drivers: how to set them up, how to write your own.

jhaack, 2015-10-02:
VIP discussion. How does it work and what can I do with it?

jhaack, 2015-10-02:
Security: How do I use the security features of VOLTTRON to protect my deployment. Also, how do I temporarily disable them when I am exploring VOLTTRON or doing local development.

jhaack, 2015-10-02:
Agent Development walkthrough: Use the ListenerAgent as a jumping off point to discuss agent development and the utilities that support it.

craig8, 2015-10-02:
A discussion of how to use the VolttronCentral agent to call a function on a remote platform.  As well as using the web api through python scripts.

MKSahm, 2015-10-02:
VIP discussion. How does it work and what can I do with it?

sumanth13, 2015-10-03:
VIP Discussion. 
how to use it with OpenADR and DR agent

jhaack, 2015-10-12:
VOLTTRON Applications: What are people doing with VOLTTRON?

closed 2016-01-29 by mikeroup