VOLTTRON/volttron #195: SQLite historian fails

jhaack, 2015-10-02:
Traceback (most recent call last):
  File "/home/rtu/git/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 168, in method
    return method(*args, **kwargs)
  File "/home/rtu/.volttron/agents/e0f3eae9-c221-4ca1-a14f-54bb8402f849/platformagentagent-0.1/platformagent/agent.py", line 341, in route_request
    **params).get()
  File "/home/rtu/git/volttron/env/local/lib/python2.7/site-packages/gevent/event.py", line 242, in get
    raise self._exception
RemoteError: AttributeError("'SqlLiteFuncts' object has no attribute 'connect'")

jhaack, 2015-10-02:
Fixing the connect issue still does not fix the historian. I am replacing with the version of the functs from the 3.x branch.

jhaack, 2015-10-03:
Fixed underlying issue. Please review.

craig8, 2015-10-08:
Reviewed and looked good

closed 2015-10-08 by craig8