VOLTTRON/volttron #197: Can't run any agent on volttron-3.0

sumanth13, 2015-10-06:
Every time I start an agent it says operation timed out. Can anyone comment on this issue please.
Below is the attached screen shot of the terminal.
Thanks in advance. 
[image]

hashstat, 2015-10-06:
The error indicates to me that either the volttron is not running or it is using a different socket address. Try running `netstat -xlp | grep vip\.socket`, which will list all the listening VIP IPC sockets associated with volttron processes. If you see no output, no volttron instances are running and, therefore, there is no VIP socket available to connect to. If any sockets are listed, you can try talking to that instance using the --vip-address option: `volttron-ctl --vip-address ipc://<ADDRESS> status`.

kmonson, 2015-10-06:
I'm having a similar problem when I run my agents by hand. They can't talk
to each other and I know the sockets are setup correctly.

Could this be related to the auth stuff?

hashstat, 2015-10-06:
This might be related to #109. Which branch? If you are on master, you can try `volttron-ctl --show-config status` and provide the results here. Equally useful would be the output of `volttron --show-config`.

craig8, 2015-10-08:
@hashstat I worked with @kmonson yesterday and determined that running the platform with the --developer-mode allowed the connection to go through.  All the connections were through local socket connections so I thought that auth would not be an issue.  Seems like perhaps it might be.  

Oh one other thing that we did (which did not help) was to zero out the curve.key file.  This did not effect the issue as far as we know.  

The combination of zeroing out the curve.key file and running in developer-mode did work.  

We did not try a non-zeroed curve.key file **and** --developer-mode to see which result that would produce.

sumanth13, 2015-10-08:
This is what i got when I tried the above commands

[image]

hashstat, 2015-10-08:
@sumanth13: @kmonson's issue was that he had an older version of zeromq which did not properly support authentication. Updating his install, zeromq in particular, resolved the issue. Can you post here the output of `pip list`? pyzmq 14.7 or later is required. Running `python bootstrap.py -u` will update all the dependencies. Thanks.

sumanth13, 2015-10-12:
In the pip list it can be seen that pyzmq 14.7 version and in the next screenshot I have updated the bootstrap.py  then the last sentence says that "Successfully installed volttron-2.0". Can you please help me with this. I am not able to run any agent till now..

[image]

[image]

hashstat, 2015-10-12:
@sumanth13 , that "operation timed out" error indicates that the control utility is unable to connect to a running instance of volttron. This usually occurs when volttron is not running or when the IPC address has been changed, such as through command-line options or a configuration file. It looks like your configuration is okay. Please try the following steps to help isolate the issue.

First, open a terminal, change to the volttron working directory, and activate the environment:
```bash
[~]$ cd volttron
[volttron]$ . env/bin/activate
(volttron)[volttron]$ 
````

Next, check to see if any instances of volttron are running and kill them:
```bash
(volttron)[volttron]$ pgrep -a volttron
46646 /home/hashstat/volttron/env/bin/python2.7 /home/hashstat/volttron/env/bin/volttron -vv
(volttron)[volttron]$ pkill volttron
(volttron)[volttron]$ pgrep -a volttron
(volttron)[volttron]$ 
```

If you get an error like `pgrep: command not found`, install the **procps** package using `sudo apt-get install procps`, which will also install *pkill*.

Then, run volttron with debugging and monitoring enabled in the terminal where you can keep an eye on it:
```bash
(volttron)[volttron]$ volttron -vv --monitor
2015-10-12 09:55:59,554 () volttron.platform.main DEBUG: open file resource limit increased from 1024 to 4096
2015-10-12 09:55:59,554 () volttron.platform.main DEBUG: using key file /home/hashstat/.volttron/curve.key
2015-10-12 09:55:59,555 () volttron.platform.main INFO: public key: 'LjGo2MhhF:4SB*dy<WSixTE3z9$Z?&]$yl/IZ72X' (kvMRapX6HJMOcERKbHwoAWi06cYfEUKA9b37lYsjM1k)
2015-10-12 09:55:59,566 () vip.monitor INFO: LISTENING 45 ipc://@/home/hashstat/.volttron/run/vip.socket
```

Now, open and activate another terminal to check the status of the new volttron instance:
```bash
[~]$ cd volttron
[volttron]$ . env/bin/activate
(volttron)[volttron]$ volttron-ctl -vv status
  AGENT                  TAG STATUS
8 listeneragent-0.1                
2 listeneragent-3.0                
7 multibuildingagent-0.1           
(volttron)[volttron]$ 
```

Finally, switch back to the first terminal, hit Ctrl-C to terminate volttron, and then copy and paste everything into a response to this issue.
```bash
[~]$ cd volttron
[volttron]$ . env/bin/activate
(volttron)[volttron]$ pgrep -a volttron
46646 /home/hashstat/volttron/env/bin/python2.7 /home/hashstat/volttron/env/bin/volttron -vv
(volttron)[volttron]$ pkill volttron
(volttron)[volttron]$ pgrep -a volttron
(volttron)[volttron]$ volttron -vv --monitor
2015-10-12 09:55:59,554 () volttron.platform.main DEBUG: open file resource limit increased from 1024 to 4096
2015-10-12 09:55:59,554 () volttron.platform.main DEBUG: using key file /home/hashstat/.volttron/curve.key
2015-10-12 09:55:59,555 () volttron.platform.main INFO: public key: 'LjGo2MhhF:4SB*dy<WSixTE3z9$Z?&]$yl/IZ72X' (kvMRapX6HJMOcERKbHwoAWi06cYfEUKA9b37lYsjM1k)
2015-10-12 09:55:59,566 () vip.monitor INFO: LISTENING 45 ipc://@/home/hashstat/.volttron/run/vip.socket
2015-10-12 09:59:29,572 () vip.monitor INFO: ACCEPTED 48 ipc://@/home/hashstat/.volttron/run/vip.socket
2015-10-12 09:59:29,572 () volttron.platform.auth INFO: 'vip' 'localhost:1000:1000:46836' 'NULL' []
2015-10-12 09:59:29,594 () vip.monitor INFO: DISCONNECTED 48 ipc://@/home/hashstat/.volttron/run/vip.socket
^CKeyboardInterrupt
2015-10-12 10:04:33,917 () volttron.platform.main INFO: SIGINT received; shutting down
Shutting down.
(volttron)[volttron]$
```

If all goes well, you should see output similar to the previous two examples. If you continue to see failures, hopefully what you paste here will provide enough information to guide you toward a solution.

sumanth13, 2015-10-12:
This is the screenshot
[image]

hashstat, 2015-10-12:
That looks normal. Did the `volttron-ctl status` command return okay? Or was there a timeout error?

sumanth13, 2015-10-12:
it worked well when i used volttron-ctl status . There was no timeout error

sumanth13, 2015-10-12:
Sorry there is a timeout error when I use volttron-ctl status now. I just tried installing weather agent and then without starting the agent i tried checking the status. It says timeout error.

hashstat, 2015-10-12:
And what happened in the other terminal, where you are running volttron? Did it crash? Was it running? Any errors?

sumanth13, 2015-10-12:
I tried using a single terminal (general way of running an agent). I am now reinstalling the volttron 3.x again. I will let you know in sometime .

sumanth13, 2015-10-12:
Thanks alot for the support. Now it is working when I reinstalled the volttron-3.x.

hashstat, 2015-10-12:
Glad to hear it. Let us know if you run into anything else.

closed 2015-10-12 by hashstat

Omnihiko, 2016-02-02:
I'm having the same issue on my build. I've gone through this write-up trying to pinpoint the error, and the following screen shot is the result of hashstat's test. Any and all help trying to pinpoint this error would be GREATLY appreciated.

[image]

craig8, 2016-02-02:
@Omnihiko which version of the source code are you using?  From the command line if you execute 

```
git branch
```
You should be able to determine the branch you are on.

Omnihiko, 2016-02-02:
@craig8 I am using the latest volttron3.0, re-installed just this morning. Using that command tells me that I'm on the master branch.

craig8, 2016-02-03:
@Omnihiko it seems like you are running the listener agent and that is where the issue is.  When you say you did a re-install can you explain what you did?

There are files under you home directory ~/.volttron that would not get cleaned up and that may be the cause of the issue.  I am doing a fresh install with the master branch that you have to see if I get the same result.

craig8, 2016-02-03:
From a clean instance of volttron I did the following operations (from my history)
```
 8  git clone https://github.com/VOLTTRON/volttron
    9  cd volttron/
   10  python bootstrap.py 
   11  . env/bin/activate
   12  volttron -vv -l volttron.log&
   13  volttron-ctl status
   14  ./scripts/core/make-listener
   15  tail -f volttron.log 
   16  history
```
Number 15 allows me to watch the log as it grows.  The listener agent creates a heartbeat every x seconds.  The following is an example of that log.

```
2016-02-03 10:07:51,727 (listeneragent-3.0 29178) listener.agent DEBUG: Peer: 'pubsub', Sender: '006cfc2e-d620-4085-9856-c08707520c94':, Bus: u'', Topic: 'heartbeat/listeneragent', Headers: {'Date': '2016-02-03 18:07:51.724942Z', 'AgentID': 'listener1', 'Content-Type': 'text/plain'}, Message: '2016-02-03 18:07:51.724942Z'
2016-02-03 10:07:56,728 (listeneragent-3.0 29178) listener.agent DEBUG: Peer: 'pubsub', Sender: '006cfc2e-d620-4085-9856-c08707520c94':, Bus: u'', Topic: 'heartbeat/listeneragent', Headers: {'Date': '2016-02-03 18:07:56.723132Z', 'AgentID': 'listener1', 'Content-Type': 'text/plain'}, Message: '2016-02-03 18:07:56.723132Z'
2016-02-03 10:08:01,725 (listeneragent-3.0 29178) listener.agent DEBUG: Peer: 'pubsub', Sender: '006cfc2e-d620-4085-9856-c08707520c94':, Bus: u'', Topic: 'heartbeat/listeneragent', Headers: {'Date': '2016-02-03 18:08:01.720755Z', 'AgentID': 'listener1', 'Content-Type': 'text/plain'}, Message: '2016-02-03 18:08:01.720755Z'
```

Omnihiko, 2016-02-04:
@craig8 Thanks for the list! Somehow I'm now able to install agents again.

For the sake of getting the data out there, I cleaned the working directory out with "rm -rf" before going through the list above. Line 13 lead to this next shot, except for the bottom instance of listeneragent (542)

[image]

Line 14 threw an error of having a previous instance, stopped that, then *successfully* installed and started the agent. Unfortunately, I didn't grab a screen cap of this.

Now I can start my own agent. I'm flummoxed as to why, but I'll take it. Thanks very much for the help!