VOLTTRON/volttron #200: Unknown subsystem 'peerlist'

craig8, 2015-10-08:
I was using the simple forwarder agent and the following happened.  The error unknown subsystem peerlist is new to me.

```
2015-10-07 13:33:33,326 () volttron.platform.main WARNING: insecure mode on key file
2015-10-07 13:33:33,326 () volttron.platform.main WARNING: empty key file; VIP encryption is disabled!
2015-10-07 13:33:53,525 (economizeragent-0.1 16556) <stderr> ERROR: /home/volttron/git/volttron/volttron/platform/agent/base.py:478: DeprecationWarning: volttron.platform.agent.BaseAgent is deprecated in favor of volttron.platform.vip.agent.Agent and will be removed in a future version. Please update agents to the new version.
2015-10-07 13:33:53,525 (economizeragent-0.1 16556) <stderr> ERROR:   super(PublishMixin, self).__init__(**kwargs)
2015-10-07 13:33:53,550 (economizeragent-0.1 16557) <stderr> ERROR: /home/volttron/git/volttron/volttron/platform/agent/base.py:478: DeprecationWarning: volttron.platform.agent.BaseAgent is deprecated in favor of volttron.platform.vip.agent.Agent and will be removed in a future version. Please update agents to the new version.
2015-10-07 13:33:53,550 (economizeragent-0.1 16557) <stderr> ERROR:   super(PublishMixin, self).__init__(**kwargs)
2015-10-07 13:33:53,562 (economizeragent-0.1 16558) <stderr> ERROR: /home/volttron/git/volttron/volttron/platform/agent/base.py:478: DeprecationWarning: volttron.platform.agent.BaseAgent is deprecated in favor of volttron.platform.vip.agent.Agent and will be removed in a future version. Please update agents to the new version.
2015-10-07 13:33:53,562 (economizeragent-0.1 16558) <stderr> ERROR:   super(PublishMixin, self).__init__(**kwargs)
2015-10-07 13:33:53,600 (simpleforwarderagent-0.1 16559) <stderr> ERROR: Generating LALR tables
2015-10-07 13:34:06,551 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 13:34:19,552 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 13:37:34,875 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 13:37:40,849 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 13:40:38,801 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 13:41:26,824 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 14:45:59,215 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 14:45:59,999 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 14:46:00,000 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 14:46:00,008 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 14:46:00,014 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 14:46:00,017 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 14:46:07,157 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 14:46:07,343 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 14:46:07,450 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 14:46:07,456 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-07 14:46:09,020 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-08 08:41:29,978 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-08 08:41:37,826 (simpleforwarderagent-0.1 16559) volttron.platform.vip.agent.core ERROR: peer u'' requested unknown subsystem 'peerlist'
2015-10-08 10:04:16,024 (datapublisher-0.1 16887) <stderr> ERROR: Traceback (most recent call last):
2015-10-08 10:04:16,082 (datapublisher-0.1 16887) <stderr> ERROR:   File "/home/volttron/git/volttron/env/local/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
2015-10-08 10:04:16,082 (datapublisher-0.1 16887) <stderr> ERROR:     result = self._run(*self.args, **self.kwargs)
2015-10-08 10:04:16,082 (datapublisher-0.1 16887) <stderr> ERROR:   File "/home/volttron/git/volttron/volttron/platform/vip/agent/core.py", line 120, in _loop
2015-10-08 10:04:16,082 (datapublisher-0.1 16887) <stderr> ERROR:     method(*self.args, **self.kwargs)
2015-10-08 10:04:16,083 (datapublisher-0.1 16887) <stderr> ERROR:   File "/home/volttron/.volttron/agents/09aba32e-3829-4541-aec3-46d6f7676974/datapublisher-0.1/datapublisher/publisher3.py", line 335, in publish_data_or_heartbeat
2015-10-08 10:04:16,083 (datapublisher-0.1 16887) <stderr> ERROR:     sensor_name, value)
2015-10-08 10:04:16,083 (datapublisher-0.1 16887) <stderr> ERROR:   File "/home/volttron/.volttron/agents/09aba32e-3829-4541-aec3-46d6f7676974/datapublisher-0.1/datapublisher/publisher3.py", line 299, in publish_point
2015-10-08 10:04:16,083 (datapublisher-0.1 16887) <stderr> ERROR:     headers=headers).get(timeout=2)
2015-10-08 10:04:16,083 (datapublisher-0.1 16887) <stderr> ERROR:   File "/home/volttron/git/volttron/env/local/lib/python2.7/site-packages/gevent/event.py", line 233, in get
2015-10-08 10:04:16,083 (datapublisher-0.1 16887) <stderr> ERROR:     result = self.hub.switch()
2015-10-08 10:04:16,084 (datapublisher-0.1 16887) <stderr> ERROR:   File "/home/volttron/git/volttron/env/local/lib/python2.7/site-packages/gevent/hub.py", line 338, in switch
2015-10-08 10:04:16,084 (datapublisher-0.1 16887) <stderr> ERROR:     return greenlet.switch(self)
2015-10-08 10:04:16,084 (datapublisher-0.1 16887) <stderr> ERROR: Timeout: 2 seconds
2015-10-08 10:04:16,113 (datapublisher-0.1 16887) <stderr> ERROR: <Greenlet at 0x7f8e1eada550: <bound method Periodic._loop of <volttron.platform.vip.agent.core.Periodic object at 0x7f8e1e840b50>>(<bound method DataPub.publish_data_or_heartbeat of)> failed with Timeout
2015-10-08 10:04:16,113 (datapublisher-0.1 16887) <stderr> ERROR: 
```

hashstat, 2015-10-08:
peerlist is a new subsystem for the router to communicate peer arrivals and departures to other peers. It looks as though the router this agent is connected to has pulled in the support while the agent has not. Those 'unknown subsystem' error messages can be safely ignored. Updating the code where the agent resides should resolve the issue.

closed 2015-10-08 by hashstat

craig8, 2015-10-08:
Updating the code to master did fix this issue.