VOLTTRON/volttron #208: Add VIP authorization support

hashstat, 2015-10-16:
Now that authentication is supported, capabilities and roles can be assigned to authentication tokens. RPC methods should be exported by the auth service to allo these capabilities and roles to be retrieved by agents using the user-id attached to VIP messages. The capabilities and roles can then be used to authorize access to RPC calls and other subsystem operations. Decorators can also be created to easily mark the access level of exported RPC methods.

I recommend a default deny policy. Credentials could be cached for a short time to improve efficiency.

zbeech, 2015-12-15:
1. Authorization should be as simple as possible on the same platform instance (server).
  * Public key should be provided from server to client via an API (TBD)

fstshrk, 2015-12-15:
Security threats that we are defending against on a single platform are:

1) Agents can not spoof each other/
2) Agents can not listen or write to topics that they are not authorized

craig8, 2015-12-15:
volttron discovery protocol

zbeech, 2016-10-17:
Authorization has been merged in for a while now.

closed 2016-10-17 by zbeech