VOLTTRON/volttron #216: Historians should use timestamps with no timezone to avoid dbapi2 bug

jhaack, 2015-10-28:
The bug in #174 has been worked around but #91 will still affect those using Python 2.7.3. The way to avoid this is to use timestamps without timezones for storage into the database.

jhaack, 2016-02-29:
Currently this is documented here: https://github.com/VOLTTRON/volttron/wiki/FAQ#id3

kmonson, 2016-02-29:
A better solution is to stop using python sqlite3 datetime parsing terribleness. 

I think we can do this without breaking existing sqlite3 historians (and base historians) by registering our own datetime parser regardless of python version. We probably don't need to create a new converter just a new parser that will use dateutil parse when we grab it back out.

kmonson, 2016-02-29:
https://docs.python.org/2/library/sqlite3.html#converting-sqlite-values-to-custom-python-types

We can replace the existing parser with our own.

kmonson, 2016-03-23:
Fixed in 7d803b3ec3f076d7809955160b0fd986d81461fc

Will make it into develop soon.

closed 2016-03-23 by kmonson