VOLTTRON/volttron #228: Show status of platforms in side frame on VC pages

jchap-pnnl, 2015-11-17:


jchap-pnnl, 2015-12-03:
# The call to get platform statuses will return a list of objects the following structure: 
{
    "result": [{
        "platform.uuid": "0987fedc-65ba-43fe-21dc-098765bafedc",
        "status": "GOOD"
    },
    {
        "platform.uuid": "2291fedc-65ba-43fe-21dc-098765bafedc",
        "status": "BAD"
    },
    {
        "platform.uuid": "4837fedc-65ba-43fe-21dc-098765bafedc",
        "status": "UNKNOWN"
    }]
}

jchap-pnnl, 2015-12-19:
We need calls to get statuses of platform agents, platform devices, and their points, so we can get statuses of children as we drill into the panel tree.

To get the platform agent statuses, use the following call:

POST /jsonrpc
{
    "jsonrpc": "2.0",
    "method": "platforms.uuid.0987fedc-65ba-43fe-21dc-098765bafedc.receive_agent_statuses",
    "authorization": "someAuthorizationToken",
    "id": #
}

The call will return a list of objects with the following structure: 
{
    "result": [{
        "agent.uuid": "0987fedc-65ba-43fe-21dc-098765bafedc",
        "agent.name": "SqlHistorian",
        "status": "GOOD"
    },
    ....]
}

To get the platform device statuses, use the following call:

POST /jsonrpc
{
    "jsonrpc": "2.0",
    "method": "platforms.uuid.0987fedc-65ba-43fe-21dc-098765bafedc.receive_device_statuses",
    "authorization": "someAuthorizationToken",
    "id": #
}

The call will return: 
{
    "result": [{
        "device.uuid": "0987fedc-65ba-43fe-21dc-098765bafedc",
        "device.name": "RTU2",
        "status": "GOOD"
    },
    ....]
}

To get the device points, use the following call:

POST /jsonrpc
{
    "jsonrpc": "2.0",
    "method": "device.uuid.0987fedc-65ba-43fe-21dc-098765bafedc.receive_device_points",
    "authorization": "someAuthorizationToken",
    "id": #
}

The call will return:
{
    "result": [{
        "point.uuid": "0987fedc-65ba-43fe-21dc-098765bafedc",
        "point.name": "CoolingValvePosition",
        "status": "GOOD"
    },
    ....]
}

craig8, 2015-12-19:
We haven't finished this yet as the calls aren't there yet

jchap-pnnl, 2015-12-19:
Clicked the wrong button to comment.

jchap-pnnl, 2016-01-08:
I've updated the general_definition.json to show statuses and other information for populating the platforms status panel.

[general_definition.json.txt](https://github.com/VOLTTRON/volttron/files/83352/general_definition.json.txt)

jchap-pnnl, 2016-01-08:
Based on our latest discussions, the tree will have the following structure:

-- platforms
    -- points
    -- agents
    -- buildings
        -- points
        -- devices
            -- points
            -- devices

jchap-pnnl, 2016-02-03:
Modified tree structure:

* platforms
 * topics
 * agents
 * sites
    * points
    * buildings
      * points
      * devices
        * points
        * devices

jchap-pnnl, 2016-02-03:
Example of above structure:

* Main-VC
  Points
  * CpuIdleTime
  * CpuIoWait
  Agents
  * ListenerAgent-3.0
  * SqlHistorian-3.0.1
  * VoltronCentralAgent-3.0
* WSU
  Points
  * OutdoorAirTemperature
  Buildings
  * BSEL
    Points
    * LightingStatus
    * WholeBuildingElectricity
    Devices
    * AHU
      Points
      * DuctStaticPressureSetPoint
      Devices
      * Zone
        Points
        * TerminalBoxDamperCommand
        * ZoneTemperature
  Agents
  * PlatformAgent
  * SQL-Historian

closed 2016-05-10 by craig8