VOLTTRON/volttron #230: Incorporate feedback on UI for registering platforms and removing agents

jchap-pnnl, 2015-11-17:
Adjust the layout in the registering platforms dialog box so the preview and buttons are visible without scrolling. (Eliminate the scrolling.)

Style the preview box so it doesn't look editable.

Put the address field next to the protocol selector.

Don't have a remove button next to the VC agent.

craig8, 2015-11-18:
Ajust the layout so that the serverkey, public and secret are all the same length and the minimum length should be what we get from the output of volttron-ctl keypair.  Two examples of that are

craig8, 2015-11-19:
I thought I would post what the new registration page looks like.
[image]

jchap-pnnl, 2015-11-19:
$$RELEASE_NOTE

UI updates to remove agent and register platform

NOTES

To remove an agent from a platform, click the Remove button in the row for that agent and confirm the removal. 

In the register-platform form, there are separate fields for the protocol, address, and encryption and authentication keys. The platform can be registered without any keys, with a server key only, or with all three -- a server key, public key, and secret key. The full concatenation of the protocol, address, and keys appears in a preview field as the values are entered.

closed 2016-10-26 by craig8