VOLTTRON/volttron #234: Error inserting duplicate topic name

jhaack, 2015-11-24:
Historian attempts to insert a topic name that already exists. It should be in the topic map at this point.

  File "/home/volttron/.volttron/agents/e8b09ae9-3df3-4555-a83f-1c7971f30377/sqlhistorianagent-3.0.1/sqlhistorian/db/basedb.py", line 94, in insert_topic
    self.__cursor.execute(self.insert_topic_query(), (topic,))
  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/mysql/connector/cursor.py", line 507, in execute
    self._handle_result(self._connection.cmd_query(stmt))
  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/mysql/connector/connection.py", line 722, in cmd_query
    result = self._handle_result(self._send_cmd(ServerCmd.QUERY, query))
  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/mysql/connector/connection.py", line 640, in _handle_result
    raise errors.get_exception(packet)
IntegrityError: 1062 (23000): Duplicate entry 'datalogger/devices/Airside_RCx/campus/building/rtu3/Supply-air T' for key 'topic_name'

jhaack, 2015-11-24:
This is caused when the capitalization on a topic name changes. The database lookup for topic names is case insensitive while the dictionary of the topics in the historian datamap is not. This causes the historian to fail to match the topic, attempt to insert, which fails due to the duplicate entry.

The suggested fix is to make the insert a replace and make it understood that topics are case insensitive in the database.

jhaack, 2015-11-24:
MYTOPIC is inserted the first time
Driver config changes the point to mytopic
Historian topic map contains "MYTOPIC" so match with mytopic fails.
Historian attempts to insert mytopic but database is case insensitive so this is a duplicate entry
If db statement is REPLACE instead of INSERT the new capitalization replaces the old but topic id is preserved.

craig8, 2015-12-02:
This is taken care of in pull request #245, however that also includes other code that isn't as specific as this task.

commit a3531160 (2016-03-11): Code Fixes:
  #234 Changed test case  test_topic_name_case_change() based on expected behavior. Marked with xfail for now as it is failing for mysql always and failing sometimes for sqlite
Clean up: formatted code based on PEP 8 requirements
	removed unused function parameters and local variables
	line length <= 79
closed 2016-10-26 by craig8