VOLTTRON/volttron #260: Packet Capture and Retrieval for Debugging

fstshrk, 2015-12-10:
In order to improve visibility into VOLTTRON behavior, as a user, I would like to have packet capture capability that can be turned ON and OFF via VOLTTRON Central for each VOLTTRON platform.

Acceptance Criteria:
1. Packet capture can be turned ON and OFF from VOLTTRON Central on a per VOLTTRON basis.
2. Packet capture is done into PCAP files of manageable size for a given platform.
3. Each capture file is timestamped and then exported to VOLTTRON Central. After export, the file is deleted from local data store.

If you have any questions, please talk to Bora/

acedrew, 2017-06-06:
I've been thinking about this same use case, wasn't going to do it through Volttron, but that may not be the worst idea. I might be willing to tackle this.

rob-calvert, 2017-06-06:
@acedrew When I worked earlier this year on a message debugging feature (services/core/MessageDebuggerAgent), I implemented unfiltered capture of the VOLTTRON bus’s message stream, but my agent didn’t support an ability to replay the packets. One reason is that, perhaps unwisely, I chose to transform the packet contents into something more easily “consumable” before storing, filtering and reporting on them. MessageDebuggerAgent does receive each message in its original format as it’s routed, though, via its own separate pub/sub socket. If you’d like to revise and repurpose any of what I did, you’re welcome to do so, and I’d be happy to answer any questions.

Rob Calvert, Kisensum

acedrew, 2017-06-07:
@rob-calvert we had an interesting discussion today about creating a general purpose inspector agent that could allow several different environmental and internal data gathering functions to facilitate on demand troubleshooting and observability.