VOLTTRON/volttron #278: Mysql version discrepancy between redhat and ubuntu based systems.

craig8, 2016-01-08:
The version in testing and the mysql historian expect the revision number of mysql to be an integer rather than a string.

On mint my version is 5.5.27-ubuntu rather than 5.5.5 as it is on red hat.

craig8, 2016-01-08:
This was fixed in 6c40816d39b372ed5d13cb08727a4c015c4ec73f 

See #277 pull request

schandrika, 2016-01-08:
@craig8  We can make the fix generic instead of ubuntu specific. I just found out that there could have suffix such as -debug and -log. We should throw away everything after the first three version number parts (everything after the first int.int.int )  as we only care to see if it greater than 5.6.4 
For default mysql installations we could use  SHOW VARIABLES LIKE 'innodb_version';
+----------------+--------+
| Variable_name  | Value  |
+----------------+--------+
| innodb_version | 5.6.27 |
+----------------+--------+
From mysql 5.1.68 this is the same as the mysql version but it is possible for someone to install a different version of the innodb plugin after the initial install.

craig8, 2016-01-08:
@schandrika I believe that's what my fix does.please verify

craig8, 2016-03-16:
-------------------------+-------------------------+
| Variable_name           | Value                   |
+-------------------------+-------------------------+
| innodb_version          | 5.5.47                  |
| protocol_version        | 10                      |
| slave_type_conversions  |                         |
| version                 | 5.5.47-0ubuntu0.14.04.1 |
| version_comment         | (Ubuntu)                |
| version_compile_machine | x86_64                  |
| version_compile_os      | debian-linux-gnu        |
+-------------------------+-------------------------+

The tests in this case are thinking that MICROSECONDS are available but they really aren't

schandrika, 2016-03-21:
@craig8 - Unable to reproduce this.

schandrika, 2016-03-24:
Unable to reproduce it on my mint VM with: 

mysql> select version();
+-------------------------+
| version()               |
+-------------------------+
| 5.5.47-0ubuntu0.14.04.1 |
+-------------------------+
1 row in set (0.00 sec)

**Console output from py.test**

** test_exact_timestamp **
('now is ', '2016-03-24T07:23:34.024930Z')
('Query Result', {'values': [['2016-03-24T07:23:34.000000+00:00', 55.76647664624326]], 'metadata': {'units': 'F', 'tz': 'UTC', 'type': 'float'}})
**('MICROSECOND SUPPORT ', False)**
('TIMESTAMP with microseconds ', '07:23:34.024930')
('TIMESTAMP without microseconds ', '07:23:34')
.deleted test records from data
In teardown method of module
closed connection to db
  AGENT                   TAG STATUS
Shutting down instance: /tmp/tmpOqwHur
shutting down platform: PIDS: [18126]
ending pid: 18126
could not kill: 18126 

======================================================================== 178 tests deselected by "-m 'dev'" =========================================================================
1 passed, 178 deselected in 2.37 seconds

craig8, 2016-05-05:
Closing as unable to reproduce again.

closed 2016-05-05 by craig8