VOLTTRON/volttron #284: Solutions to reduce security concerns from Facility Managers

loulupe, 2016-01-12:
We would like to connect Volttron to our existing building automation system (BAS - Automated Logic).  There is a reasonable concern from the Facility Management office about the interactions of Volttron to the BAS and campus network (unintentionally shutting down or turning on devices, or other vulnerabilities).  

We would appreciate recommendations or solutions that would give peace of mind to the Facility Managers which would facilitate greater access for researchers.  If possible options that they could control from their existing system.

Thanks

jhaack, 2016-01-12:
I have started a page at: https://github.com/VOLTTRON/volttron/wiki/Recommendations-for-Deployments

loulupe, 2016-01-14:
Thanks!LG

Date: Tue, 12 Jan 2016 13:23:00 -0800

closed 2016-10-26 by craig8