VOLTTRON/volttron #291: Improve Base Historian Performance

kmonson, 2016-01-13:
On low power devices the base historian has trouble keeping up with a large volume of messages. 

We need to improve how we use sqlite or replace it with something else.

kmonson, 2016-01-19:
Still other improvements to be made. There is an index on the cached history table that is probably unneeded.

kmonson, 2016-01-19:
Could also experiment with increasing page sizes.

craig8, 2016-10-26:
@kmonson update or close if we are good with our current.

kmonson, 2016-10-26:
We still need to look at this.

kmonson, 2016-11-23:
I looked into removing an index that is not technically required but does add some safety thinking that it would help with performance. It did not. As that is the case we can safely assume that changing page counts in sqlite will not be beneficial as well.

closed 2016-11-23 by kmonson