VOLTTRON/volttron #301: Modbus and BACnet support over RS-485

jhaack, 2016-01-21:


craig8, 2016-10-26:
Thoughts?  Look into our EVSE work for thoughts perhaps?  Is this doable and do we want to do it now. or future?

kmonson, 2016-10-26:
BACnet and Modbus over serial are a very different animal that firing up pyserial and giving it a go (which is what out EVSE stuff basically does). 

BACnet is a complex protocol and very sensitive to timing issues. This would require rewriting the bacnet proxy (or writing a separate special one) from scratch. It would be probably a months worth of work (or more). Also, in a non-realtime OS (such as Linux) BACnet over serial is fraught with peril. I can't say for certain it would work.

jhaack, 2018-05-30:
Closing until community need

closed 2018-05-30 by jhaack