VOLTTRON/volttron #303: BACnet driver expanded to support BACnet schedules

jhaack, 2016-01-21:


jhaack, 2016-01-27:
From user request:

"I wanted to bring up the idea of needing the final trending / logging solution to be able to also collect BACnet TL (Trend log) objects as well as trend present value.  I know currently the Volttron BACnet driver cannot do this.  Reasoning - we see some sites, and more in the future that have many or all points already trended in the native BAS.  Adding a Volttron instance locally that duplicates this would increase local traffic and two devices would be trending.  So, the better solution is to collect the data from the local TL object and forward to central data base.  This collection could run once per day, or every 4 to 8 hours depending on the need for near real time data or the local trends rollover timing. "

craig8, 2016-10-26:
@kmonson please comment on the feasibility and perhaps how one would extend the BACNet driver.

schandrika, 2020-02-14:
@kmonson Did you remember checking the feasibility on this?