VOLTTRON/volttron #312: Reconnecting platforms

carlatpnl, 2016-01-27:
scenario:  platform 1: raspberry pi, running a 'Hardware Agent'
                platform 2: laptop, running a 'Control Agent'
both agents talk VIP to each other.

issue:  shutdown platform 2
            restart platform 2
            agents can no longer communicate:
            platform 1 reports "exception: <class 'volttron.platform.vip.socket.ProtocolError>" when trying
            to send to platform 2

carlatpnl, 2016-01-27:
also, if i shutdown platform 2 and leave platform 2 running, when i restart platfrom 2 communication resumes fine.

craig8, 2016-02-24:
This was present on the vcollab vc demonstration code.  During security updates.

closed 2016-05-07 by craig8