VOLTTRON/volttron #328: VOLTTRON device data backup

utalk2, 2016-02-08:
Need platform to store hardware/software device data in a package and provide all files required to duplicate a VOLTTRON instance. This could be provided for use of the VOLTTRON Management Console to backup and restore failed VOLTTRON instances.   Other front ends could request the same package as does the Management console.

craig8, 2016-10-26:
I believe that this is outside of the scope of volttron proper, however I am investigating ansible through ssh to accommodate this.

schandrika, 2020-02-14:
@craig8 could you comment on this?

schandrika, 2020-02-22:
Ansible scripts will help setup agents with the same configuration as an existing environment. Backup of volttron home or historian data is not within scope of core VOLTTRON