VOLTTRON/volttron #330: Need more BACnet device properties supported

utalk2, 2016-02-08:
Currently the platform provides device ID and manufacture code.  Need to have all BACnet properties available such as: Make, model, serial number, software version, firmware version, and name that was provided for the device (i.e. AHU-1 named by technician).

craig8, 2016-10-26:
@kmonson please comment on feasibility and such.

kmonson, 2016-10-26:
Just needs time, but much of the properties mentioned are meta data and not really stuff we want to scrape every time. Seems like this would be more suited the configuration grabbing phase. Maybe we need to add a meta data section to device configurations that gets published with other meta data.

acedrew, 2017-06-06:
This data would be useful for the various "autotagging" schemes we've been discussing.