VOLTTRON/volttron #344: Actuator: exception message sent during unhandled exception does not have a uniform format

schandrika, 2016-02-09:
When there is a LockError the RemoteError thrown has 
e.exc_info['exc_type'] = 'actuator.agent.LockError' and 
e.message == 'caller does not have this lock'

But I do not get the same format when I get a TypeError. i.e. below two assert statements fail
 assert e.exc_info['exc_type'] == 'TypeError'
 assert e.message == 'float() argument must be a string or a number'

schandrika, 2016-02-25:
There was discussion about possible use of python pickling. But I assume this is of lower priority

craig8, 2016-10-26:
@schandrika is this still an issue?  @kmonson  can you comment?

kmonson, 2016-10-26:
It is, but it's not a super high priority. 

It should be pretty easy to fix.

jklarson, 2020-07-31:
The format of the error type should match in all cases - either always "typeError/LockError" or "<thing>.Error and <thing>.Error2"

schandrika, 2021-02-20:
@jklarson does the error message clearly say that it is a lockerror. If so I think we should just fix the test case and close this bug since this is a lower priority issue