VOLTTRON/volttron #355: Documentation Clarification

carlatpnl, 2016-02-19:
https://github.com/VOLTTRON/volttron/wiki/Eclipse-Dev-Environment 
section 5 states import as a general project, then shows the Import Projects... screen shot.
eclipse does not present a selection option dialog to choose "Import existing Eclipse Project" "Import using New wizard" or "Import as genera project" until AFTER you have selected Import Projects.  This can lead to confusion, Suggest showing the option dialog with the "Import as general project" radio button selected.

craig8, 2016-10-26:
@carlatpnl Can you puruse the new documentation and see if we have de-confused this isssue for you https://volttron.readthedocs.io/en/master/search.html?q=eclipse&check_keywords=yes&area=default

acedrew, 2019-11-15:
@carlatpnl Can you complete reviewing this so the issue can be closed?

shwethanidd, 2020-02-17:
@schandrika can you check and close this issue