VOLTTRON/volttron #365: Bootstrap fails to build gevent for Python 2.7.6.

rlutes, 2016-02-25:
Bootstrap process will fail to build wheel for gevent.  This occurs for gevent > 1.0.1 and Python version 2.7.6.  Work around is to install older version of gevent or newer Python interpreter.

craig8, 2016-02-25:
I am not sure what to do with this as this is my history and everything worked out with a minimal server just like you mentioned.

See the following image.

[image]

In addition http://volttron.readthedocs.org/en/develop/ is the instructions that I used which are the same that is in the quickstart of the wiki.

Was yours on a board or on a desktop system?  VM or Not?

rlutes, 2016-02-25:
64 bit minimal install and non-server.

craig8, 2016-02-26:
Ok I will try that now.

rlutes, 2016-02-26:
Could not replicate issue on another system running the same Ubuntu version and Python interpreter.

closed 2016-02-26 by rlutes