VOLTTRON/volttron #390: Implement an optional heartbeat periodic for the base agent

jhaack, 2016-03-09:
This should be on by default for service agents. Can be used by other agents to announce they are alive.

craig8, 2016-03-10:
@jhaack @mikeroup @kmonson  please review and let me know if I have scoped this out well enough.

mikeroup, 2016-03-11:
Why is the topic 'heartbeat/agentid/uuid'? The uuid is added automatically by pubsub.

craig8, 2016-03-11:
Ummmm...are you talking about the sender as grabbing the uuid?  That is not the uuid that is the vip.identity.  The uuid I am talking about is the actual agent installation directory .volttron/agents/uuid/listeneragent-0.1.  The vip.identity (depending on which agent is started) isn't unique.

If I am incorrect then I am not sure what you mean about it being added automatically by pubsub.

mikeroup, 2016-03-11:
Okay but what about service-y agents that are started in main? They aren't explicitly installed like other agents.

craig8, 2016-03-12:
For the service agenta there will only be one on a specific instance so it will be unique.  For any agent that is "installed" i.e. via volttron-ctl there will be a uuid folder name.  There may or may not be a unique vip.identity specified in the agent configuration file.

The ultimate goal is to be able to state that at this point in time this agent known as foo is running.  If there are multiple foo agents installed there need to be a way to distinguish between each of the installed agents.  

Note you can get the agents uuid directly from the environment using os.environ['AGENT_UUID']

craig8, 2016-03-14:
What is interesting is I keep waffling about where this should be put.  Now I think it should hang off the pubsub system since it will use pubsub as it's mechanism for the heartbeat.

mikeroup, 2016-03-14:
It would use pubsub but is not directly related to its implementation. It also needs to know about rpc and core.

closed 2016-05-05 by craig8